svenhb/GRBL-Plotter #5: Some Bugs

Gabse, 2016-09-11:
Hi, 
I’ using your software for quid a while and found 2 bugs and some suggestions:
First bug:
There is a Bug at the Mirror commands, the g-code won’t machine anymore after mirroring an axis.
Second bug:
The Software will arbitrarily perform a “Kill Alarm” after running for a longer time (or maybe also if too many g-code files where loaded successively?) A restart of the software will solve the issue. It happens on manual control as well as while streaming a file.
Some suggestions:
Could you please add a field to input user defined values for the driving distances over the joystick?
Is it possible to reduce the feedrate and the spindle speed percentage (from 0-150%) like on large machines?

svenhb, 2016-09-13:
Thanks for the Feedback
I will try to fix the bugs next Weekend
Can you please explain you last whish (Is it possible to reduce the feedrate and the spindle speed percentage (from 0-150%) like on large machines?)?

Gabse, 2016-09-17:
Professional CNC milling machines normally do have two potentiometers for de-/increasing the spindle speed and the federate while the machine is processing. The reason is to test a nc file slowly or step by step and to reduce / increase federate and spindle speed additionally without recalculating the whole file. It would be very nice if you could add two software potentiometers for those reasons and maybe also link them to gamepad if that once is supported.

commit 50309bfc (2016-09-18): Vers 0204

fix #5
closed 2016-09-18 by svenhb