svenhb/GRBL-Plotter #8: slow and odd  engraving speed

mkeyno, 2017-05-16:
hi, I set the GRBL 1.1f and software  configuration as per following images , however engraving speed is not what I expected , I saw many CNC machine use high rate of reciprocating movement during the engraving job , but I haven't seen such movement for any of engraving attempt , can you please let me know whether is possible to use GRBL plotter for such high speed engraving ?
[image]
[image]

svenhb, 2017-05-17:
Hi,

I don't know if I got your question correct...

The max. speed is limited by the arduino hardware: 30kHz is the max frequency at the clk-signal for each stepper.

Check https://github.com/grbl/grbl/issues/41

 

You are way above with your settings:

126 steps/mm; 10000 mm/minute = 126*10000*60 = 75600kHz

 

Best regards

SVen

 

Gesendet: Dienstag, 16. Mai 2017 um 20:19 Uhr

mkeyno, 2017-05-17:
@svenhb are you sure about equation? shouldn't be divided by 60? however I'm looking for correct setting to reach highest engraving speed, have you perform engraving job with this software?

svenhb, 2017-05-17:
You are right, sorry... 126*10000/60 = 21kHz. 
I use my tool for plotter and small cnc control. Because of Hardware Limits my max. Speed is 4000 mm/Minute - and it works...
What is exactly your problem? Probably the generated GCode is not good for high Speed engraving?
How the gcode could be improoved? Did you find other Tools, generating gcode, which are different?

mkeyno, 2017-06-21:
@svenhb  do you have any plan to add dxf or dwg import feature to your software

svenhb, 2017-06-22:
Hi,

I had a rough look on the dxf data format, but it was too strange at that time - perhaps I will check again (perhaps there is code on github I could use).

Now I will implement rotary-axis substitution (rotary axis instead of X or Y axis) and Gcode 'wrapping' for easy switch between X(or Y)-axis and rotary axis.

My current problem: one turn of rotary axis corresponds to 1mm of X-axis and I need to scale just X-dimension of code...

 

Gesendet: Mittwoch, 21. Juni 2017 um 19:23 Uhr

mkeyno, 2017-08-03:
Hi @svenhb , I've tried to use Persian letter for engraving , but its seems it is not supported ,  can you tell me how to add this feature to your software ?
Thanks

svenhb, 2017-08-03:
Hi,

I took the character data from http://www.evilmadscientist.com/2011/hershey-text-an-inkscape-extension-for-engraving-fonts/

and converted them to c# data.

For example "Sans 1-stroke" contains this data:         new string[] {"-8 8","-12 12 M -8 -10 L -4 -8 .../ break /...... L 9 -1 L 9 -3" },

(Some 25000 characters) to decribe each single character from the "Sans 1-stroke" alphabet.

 

Just have a look in GCodefromText.cs. The Order of fontNames is the same as the charSets in "fontChar"

 

If you have such data for the Persian letter I could include it.

 

 

 

Gesendet: Donnerstag, 03. August 2017 um 12:13 Uhr

commit c0809dd5 (2017-08-09): Fix #8
closed 2017-08-09 by svenhb