svenhb/GRBL-Plotter #9: installation process in german

alfredanil, 2017-11-03:
Hallo
Q1) would it possible to change the installation setup process to english or system default language, right now it is in german? 
Q2) does it support Grbl for mega as well?
Danke

svenhb, 2017-11-03:
Hello, 
1)I set default language to english for all forms, I did'nt know setup behavior is different, I will check with next version.
2) as long as mega-GRBL version uses same GCode commands and answers with same response messages, it should work.

svenhb, 2017-11-04:
Btw. in priciple there is no installation needed, You can copy the folder "GRBL-Plotter/bin/Release" and start GRBL-Plotter.exe from this folder.

svenhb, 2017-11-07:
I'm sorry, I have no plans to do that: 3D (stl, obj) is much too complicated and not the intended function of GRBL-Plotter. I have no idea about controlling a lathe: doesn't it use the same GCode commands?

alfredanil, 2017-11-19:
yes, lathe uses the same GCode commands. Your doing a great job with this software.

closed 2017-11-19 by alfredanil