svenhb/GRBL-Plotter #10: 2d path view can not be zoomed

mkeyno, 2017-11-12:
hi @svenhb , 
I've noticed 2d path view area can not be zoomed whereas G-cod text are perfectly zoomed , is this any feature that I miss about that?
BR

svenhb, 2017-11-12:
Are you unsing latest version 1.1.2.0? Zooming is then connected to mouse-wheel when mouse-pointer is above 2d view.

mkeyno, 2017-11-13:
thanks @svenhb   I did but it seems  it works partially , and during the CAM job if feedrate or zoom changed , the program  jammed , 
do you have any plan to make some changes in your software ?

svenhb, 2017-11-13:
During a CAM job the 'realtime' 2d view will be replaced by a picture (to avoid redrawing the whole graphics with each change of the tool position) to increase performance on (my) XP PC. 
I will check if can solve the described problem.

mkeyno, 2017-11-13:
can you add  feedrate in the control area like many CAM software ?

svenhb, 2017-11-13:
Feedrate for what? When using the joystick control, feedrate is taken from setup-values.
Perhaps you have a screenshot for me as an example.

mkeyno, 2017-11-14:
I don't have joystick  and may user need decreased or increased speed of CAM job during operation  , many software like Repiteir have this option
[image]

and  when I change the feedrate in override screen , the software got jammed , 
don't you think is it better to add this feature in control/jogging area  ?
[image]

svenhb, 2017-11-14:
I meant the yellow control aerea - this is the joystick-like-control...
[image]

> feedrate in override screen.....  to add this feature in control/jogging area
I don't use it very much but I will check where I could add it to the main form.

mkeyno, 2017-11-14:
thanks @svenhb  , please let me know if have any plan to enhanced its features , I have couple of suggestion that may useful for user-friendly capability of your software

svenhb, 2017-11-14:
Did you try GRBL Version 1.1 for feedrate override? Your screenshot shows GRBL 0.9 override screen...
With GRBL 0.9 I have to interrupt the data stream to inject the new feedrate command - probably this is the reason for jamming.
With GRBL 1.1 override is handled by GRBL itself.

mkeyno, 2017-11-17:
no I've always use last available frameware , have you check the feedrade change  during CAM job?

svenhb, 2017-11-17:
With GRBL 0.9 I checked one year ago (with GRBL-Plotter 1.0? I don't remember). With 1.1 I checked 1 hour ago on my old XP-notebook and it worked.

hilam8899, 2018-01-03:
HI,@svenhb
when I try to compile grbl_plotter ,
CS2001	Source file 'D:\9\VISUAL_STUDIO_PROJECT\GRBL-Plotter-master\GRBL-Plotter\eventHandler.cs' could not be found.	GRBL-Plotter	D:\9\VISUAL_STUDIO_PROJECT\GRBL-Plotter-master\GRBL-Plotter\CSC	1	
can you help me?thank you!

closed 2018-01-03 by svenhb

svenhb, 2018-01-03:
I forgot to upload this file - now done.

commit 73901851 (2018-01-03): Fixed #10 Update 1.1.3.0