svenhb/GRBL-Plotter #13: questions

Gabse, 2018-02-23:
Hi, I recently updated to GRBL 1.1f and GRBL-Plotter 1.1.3.0, and have a few questions:

-I wanted to try out the new overwrite function, but the options are grayed out on your program, do I have to enable them first somewhere? 
-Is it possible to know the install dir/change it to C:/ Programs, because I wanted to add script files to the programmable buttons, but I can’t find the _misc folder
-Is it normal, that the software says "Laser Mode not active $32=0" on startup?
-During a file my machine hit an end stop, and GRBL aborted the process. The COM window of GRBL-Plotter reported it, but the main window still said Run, and after resetting it over the COM window on the main window there was flashing kill Alarm until I reseted it again over the GUI.
-Is it planned to add joystick support to the software, using the jogging mode? It would be very handy for probing out the workpiece
-Bei mir funktioniert die Umstellung auf Deutsch nicht, es erscheint die Meldung, dass ich die Software neu starten muss, aber nach dem Neustart ist sie immer noch englisch

svenhb, 2018-02-24:
- the override function will be enabled automatically, when Grbl 1.1 is recognized. After a reset GRBL-Plotter is looking for the string "Grbl 0" or "Grbl 1" to identify the version - perhaps your version answers in a different way? Please tell me.
- the only way I know, to select the install-dir, is to copy the the content of https://github.com/svenhb/GRBL-Plotter/tree/master/GRBL-Plotter/bin/Release to the location you like - without any installation
>-Is it normal, that the software says "Laser Mode not active $32=0" on startup?
Yes, for some one it was important and I don't care ;-)
- I need to improve the communication between the serial-form and the main GUI, some times after reset or error, the GUI is not updated...
- I don't plan to support a joystick, but if it is simple to implement I could do it. What kind of joystick do you use?
>- Language-  nach dem Neustart ist sie immer noch englisch
Yes, strange - I never tried the installed version - I only started GRBL-Plotter from the bin/Release folder up to now...
Probably some work to fix this...

Gabse, 2018-02-25:
> After a reset GRBL-Plotter is looking for the string "Grbl 0" or "Grbl 1" to identify the version - perhaps your version answers in a different way? Please tell me.

I’m using the standard version of GRBL Mega (1.1f)

> the only way I know, to select the install-dir, is to copy the the content of https://github.com/svenhb/GRBL-Plotter/tree/master/GRBL-Plotter/bin/Release to the location you like - without any installation

I will try that thanks

> I don't plan to support a joystick, but if it is simple to implement I could do it. What kind of joystick do you use?

At the moment I don’t have any joystick, but I’ve seen videos, where people use Xbox Controllers

> Yes, strange - I never tried the installed version - I only started GRBL-Plotter from the bin/Release folder up to now...
Probably some work to fix this...

Habe es gerade auf einem anderen PC ohne Installation gestartet, und auch hier dasselbe verhalten.

svenhb, 2018-02-25:
>I’m using the standard version of GRBL Mega (1.1f)

The override buttons inside the GUI are only enabled if the version is right AND gcode will be sent to the arduino. Can you show the text coming from the µC after reset - perhaps your text only shows uppercase? Usually it looks like this:
```
> RESET
Grbl 1.1f ['$' for help]
```

>Habe es gerade auf einem anderen PC ohne Installation gestartet

In der nächsten Version versuche ich eine andere Einstellung für die 'Veröffentlichung'. Vorher haben andere User berichtet, dass des Setup nur in deutsch abläuft - also hatte ich die Veröffentliche Sprache von 'default' auf English geändert - damit werden alle weiteren Sprachen (sofern verfügbar) nicht mehr exportiert.

svenhb, 2018-02-25:
I made a new version 1.1.3.1 : https://github.com/svenhb/GRBL-Plotter/blob/master/GRBL-Plotter_1131_Publish.zip  
- publishing language option 'default' - Nach regulärer Installation kann nun zwischen de und en gewechselt werden  
- Check of GRBL version case independend  
- Show override button alltime (as long as grbl 1.x was recognized)

Gabse, 2018-03-20:
Thank you for the update, it solved the problems, but unfortunately now I’ve got another problem:

When trying to check or execute a file with the feederate chanced multiple times, it starts, but then aborts, gives me the error:

```
[Start code check]
Save last pos.: X-201 Y-1 Z-1
> $C 
> (#T1 M6 M3) 
> G17 
> G0Z20.000 
> G0X0.000Y0.000S12000 
> G351 X-1 
> G0X89.956Y-89.956Z6.000 
> G1Z-0.500F100 
[MSG:Enabled]
> G1X90.000Y-90.000F2000 
> G1X89.956Y-90.044 
> G1X90.000Y-90.000 
<
< !!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
<
< error:20 	Unsupported or invalid g - code command found in block.
< Error before code line 10 
```

[image]

Any idea, what could be wrong?
And another thing: when I try to transform (rotate/scale...) the gcode, it delates the additional feedrates, leaving only identical ones standing (for example all F100 in the file above)

svenhb, 2018-03-20:
Hi, I'm not sure if read correct: isn't it "G35" in line 5?
I can't fin this code in the supported list: https://github.com/gnea/grbl/wiki#nice-features
[image]

About the 2nd problem: I try to fix it soon

Gabse, 2018-03-21:
Thank you for your reply.
Yes, the problem was the G35 comment, written by my Postprocessor. Strange, that the problem has not existed on the old GRBL version.
Just one last question: is it normal that when I try to pause or abort a file the controller continues a couple of lines until it reaches the pink point, before pausing? 

Many thanks

svenhb, 2018-03-21:
>is it normal that when I try to pause or abort a file the controller continues a couple of lines until it reaches the pink point, before pausing?

Yes because this commands are already in the command buffer of GRBL. GRBL-Plotter stops sending new commands, but GRBL needs to empty its own buffer until 'IDLE' is state is reached: https://github.com/gnea/grbl/wiki/Grbl-v1.1-Interface#streaming-protocol-character-counting-recommended-with-reservation
As I know, only a reset can emtpy and stop processing the stored commands inside the GRBL buffer.

svenhb, 2018-03-25:
Check out new pre-release, supporting a gamepad: https://github.com/svenhb/GRBL-Plotter/releases/tag/v1.1.4.0 
Unfortunately the behavior of the 'real' analog joystick is strange - I spend hours to get the best behavior. Better check by your own.

closed 2018-03-25 by svenhb