svenhb/GRBL-Plotter #14: Eagle Drill file - Potential Addition / pull request

robZeilinga, 2018-03-22:
Hi Sven, (i presume) 
fantastic work!! 
I was struggling to write my own sender in C# until i stumbled on your project. 

The use case i am trying to solve is a very easy process to drill PCB's created by Eagle (or others) - these create a *.drd file (drill file) that contains the co-ordinates (and size) of the required holes needed.
the challenge is that the file assumes that your etched PCB is mounted to the CNC table and exactly aligned to the X/Y axis, which is almost impossible.  I have done previous attempts such as positioning 4mm pins and holes on the PCB to align and exactly position the PCB, but this is a hassle. (albeit working quite well) 

Then I tried to pick up the amount of angle via a camera mounted to my cnc, and the image complete with cross-hairs will display the PCB image, selecting two holes on the Drill file display and matching holes on the PCB camera display will automatically calculate the amount of angle and offset.. (sounds familiar doesn't it?  :) ) 

but i had hit a brick wall getting the gcode to stream to the GRBL in a controlled manner. and that's when I came across your project. 

Would you consider allowing this enhancement? I don't want to simply fork and create a pull request without discussing this with you.

Regards
RobZ
[image]

[Sample_DrillFile_drd.txt](https://github.com/svenhb/GRBL-Plotter/files/1837635/Sample_DrillFile_drd.txt)

svenhb, 2018-03-22:
Hello Rob,
your suggestion sounds good to me. At least an importer for *.drd I will try to implement. 
My first point is to implement a (real) joystick control - but I still wait for my ordered gamepad...
I hope to release the next version in in one or two weeks.

svenhb, 2018-03-25:
Check out the new pre-release with Eagle drill file import - I hope it does the job...
https://github.com/svenhb/GRBL-Plotter/releases/tag/v1.1.4.0
Later on I need to figure out best practise for camera supported code transformation.

closed 2018-03-25 by svenhb