svenhb/GRBL-Plotter #15: Support for KiCad and fiducial camera

alfredanil, 2018-03-29:
Would it be possible to support Kicad drill file as well. And also fiducial camera or marker for accurate positioning.

svenhb, 2018-03-29:
>Would it be possible to support Kicad drill file as well
Please send me an example file + a screenshot - I don't have KiCad.
>And also fiducial camera or marker for accurate positioning.
Is in progress - should be available soon.

alfredanil, 2018-04-01:
Default units for KiCad is in inches. Here are some supported files for KiCad drill 
[image]
Thank you for adding Fiducial cameras and keep up the great job.

hilam8899, 2018-04-02:
Hi,svenhb:
  I want to add the 4th axis C(Rotation Z ）,having tangent to XY plane to cut wood. how can I finish my machine?Thank you

svenhb, 2018-04-02:
Do you use axis substitution, a 2nd grbl controller or a real 4th axis on one controller?
For a real 4th axis you need a special grbl version (I don't know which one, I just support status message).
Here I show axis substitution: https://youtu.be/BLRfBk9yIbw

hilam8899, 2018-04-03:
Hi,svenhb:
GRBL 0.8C support 4th,you can see it:
https://github.com/gcobos/grbl4axis
Now I use mega2560 ARDUINO .Unlike your machine,my machine have real 4th.,it is C axis ,not A, I am wondering how to have a tangent  to XY plane.
Thank you!

svenhb, 2018-04-03:
In [Setup - Control - GRBL 4th axis] you check "Show status and controls" and selecr "C" as axis.

>I am wondering how to have a tangent to XY plane.

I don't know what you mean? How doeas you machine looks like?
GRBL-Plotter will not calculate any 4th axis position or movement. It just can show 4th axis state and forward 4th axis commands.
You need to use a CAM software which can generate code for real 4th axis.

hilam8899, 2018-04-03:
FOR GRBL-Plotter having DXF import function, I want to finish CNC to cut wood or glasses,not using the other cam to create Gcode. C axis  coordinates is creating from X axis and Y axis ,I want to add a cam to GRBL-Plotter ,I found qcad perfect.

hilam8899, 2018-04-03:
Add a cam to GRBL  Plotter .Isn't that a good idea?

svenhb, 2018-04-03:
DXF import and camera are already implemented.
Check Wiki for setup: https://github.com/svenhb/GRBL-Plotter/wiki/Camera-setup

svenhb, 2018-04-07:
In 1.1.5.0 implemented

closed 2018-04-07 by svenhb