svenhb/GRBL-Plotter #17: Missing font

endstreet, 2018-05-08:
The font wqy-unicode.llf is included in the project but not included in the zip. You can just remove the font from the project as it is not used. Currently the build is failing because the font is missing.

svenhb, 2018-05-08:
The size of this font-file is 42MB, if I try to upload to github I get the message "Yowza, that’s a big file. Try again with a file smaller than 25MB." - what can I do?
Yes I should remove this font from the project.

endstreet, 2018-05-08:
You can extract it from there..since it is not used in the project you can
just exclude it. The project will build at least. Zipping these resources
is maybe an option if you need to include it for dxf functionality.

commit 0091987b (2018-05-08): fixed #17 removed font 'wqy-unicode.llf' from project
closed 2018-05-08 by svenhb