svenhb/GRBL-Plotter #18: simple send and Receive data to grbl

ashkan4818, 2018-05-19:
hello 
i want to create a simple Interface for GRBL in C# . i really tried so hard for do this but i could to get a good result .then i find you in Github I looked at your wonderful GRBL - plotter  application code and i could to understand some thing about your code , then i tired to create a simple application like you that able to receive data from GRBL and show that on a rich box  therefore  I wrote this code below 
But unfortunately it still does not work properly :( i'm really disappointed

can you help me about this issue?? i really need to your help , thank you so much <3

#########################################################################

`using System;
using System.Collections.Generic;
using System.ComponentModel;
using System.Data;
using System.Drawing;
using System.IO.Ports;
using System.Linq;
using System.Text;
using System.Threading.Tasks;
using System.Windows.Forms;
using System.IO;

namespace serial_test5
{
    public partial class Form1 : Form
    {
        public string rxString;

        public Form1()
        {
            InitializeComponent();
        }

        private void Form1_Load(object sender, EventArgs e)
        {
            string[] ports = SerialPort.GetPortNames();
            CoMports_list.Items.AddRange(ports);
            CoMports_list.SelectedIndex = 0;
            serialPort1.DataReceived+= new System.IO.Ports.SerialDataReceivedEventHandler(serialPort1_DataReceived);
        }

        private void serialPort1_DataReceived(object sender, SerialDataReceivedEventArgs e)
        {
            while ((serialPort1.IsOpen) && (serialPort1.BytesToRead > 0))
            {
                rxString = string.Empty;
                try
                {
                    rxString = serialPort1.ReadTo("\r\n");
                    this.Invoke(new EventHandler(handleRXData));
                    while ((serialPort1.IsOpen)) ;
                }
                catch (Exception exception)
                {
                    Console.WriteLine(exception.Message , "can not received data");
                   
                }
            }

        
        }

        public void addToLog(string text)
        {
            richTextBox1.AppendText(text + "\r");
            richTextBox1.ScrollToCaret();
        }

        private void handleRXData(object sender, EventArgs e)
        {
            char[] charsToTrin = {'<', '>', '[', ']', ' '};
            int tmp;

            if (rxString.IndexOf("ok") >=0)
            {

                    addToLog(string.Format("<{0}",rxString));

                return;

            }
            else if (((tmp = rxString.IndexOf('<')) >= 0) && (rxString.IndexOf('>') > tmp))
            {
                if (scroling_chek_box.Checked)
                    addToLog(rxString);
               
                return;
            }

        }

        private void Connect_btn_Click(object sender, EventArgs e)
        {

            serialPort1.BaudRate = Convert.ToInt32(baudRate_list.Text);
            try
            {
                serialPort1.PortName = CoMports_list.Text;
                serialPort1.Open();
            }
            catch (Exception er)
            {
                MessageBox.Show(er.Message, "message", MessageBoxButtons.OK, MessageBoxIcon.Error);
            }

        }

        private void timer1_Tick(object sender, EventArgs e)
        {

            if (serialPort1.IsOpen) // 
            {
                try
                {
                    var dataArray = new byte[]
                    {
                        Convert.ToByte('?')
                    }; 
                    serialPort1.Write(dataArray, 0, 1);                      
                }
                catch (Exception er)
                {

                    serialPort1.Close(); 
                }
            }

        }

    }
}
`

svenhb, 2018-05-19:
What is the problem - what does not work proberly?

ashkan4818, 2018-05-19:
i can not see the GRBL status messages in rich box , i don't know where is my wrong

ashkan4818, 2018-05-19:
this is all of thing that i can see of status messages:

[https://ibb.co/kH4nO8](url)

ashkan4818, 2018-05-19:
yes i sent a '?' to grbl to get status report! Yes I have read it...
can you write a very simple code for sending a  g-cod to grbl and read show  status report in rich box? please?

svenhb, 2018-05-19:
You bettet look for 3dp burner on github, this was my  starting point for serial communication to grbl.
-- 
Diese Nachricht wurde von meinem Android Mobiltelefon mit GMX Mail gesendet.

"@type": "MessageCard",
"@context": "http://schema.org/extensions",
"hideOriginalBody": "false",
"originator": "37567f93-e2a7-4e2a-ad37-a9160fc62647",
"title": "Re: [svenhb/GRBL-Plotter] simple send and Receive data to grbl (#18)",
"sections": [
{
"text": "",
"activityTitle": "**ASHKAN**",
"activityImage": "https://avatars3.githubusercontent.com/u/20372650?s=160\u0026v=4",
"activitySubtitle": "@ashkan4818",
"facts": [

]
}
],
{
"name": "Add a comment",
"@type": "ActionCard",
"inputs": [
{
"isMultiLine": true,
"@type": "TextInput",
"id": "IssueComment",
"isRequired": false
}
],
"actions": [
{
"name": "Comment",
"@type": "HttpPOST",
"target": "https://api.github.com",
"body": "{\n\"commandName\": \"IssueComment\",\n\"repositoryFullName\": \"svenhb/GRBL-Plotter\",\n\"issueId\": 18,\n\"IssueComment\": \"{{IssueComment.value}}\"\n}"
}
]
},
{
"name": "Close issue",
"@type": "HttpPOST",
"target": "https://api.github.com",
"body": "{\n\"commandName\": \"IssueClose\",\n\"repositoryFullName\": \"svenhb/GRBL-Plotter\",\n\"issueId\": 18\n}"
},
{
"targets": [
{
"os": "default",
"uri": "https://github.com/svenhb/GRBL-Plotter/issues/18#issuecomment-390416995"
}
],
"@type": "OpenUri",
"name": "View on GitHub"
},
{
"name": "Unsubscribe",
"@type": "HttpPOST",
"target": "https://api.github.com",
"body": "{\n\"commandName\": \"MuteNotification\",\n\"threadId\": 337051977\n}"
}
],
"themeColor": "26292E"
}

ashkan4818, 2018-05-20:
i will see this and try again . thank you so much sir <3

svenhb, 2018-05-31:
Hope you found a solution

closed 2018-05-31 by svenhb