svenhb/GRBL-Plotter #22: Verwendung eines Hardware-Joystick am GRBL-Plotter

deHarro, 2018-07-09:
Hi Sven,
zuerst: Kann ich die Idee in deutsch formulieren, oder muss alles auf Github in Englsich laufen?

Harald

svenhb, 2018-07-09:
Gute Frage, ich würde jetzt so weitermachen, da der Titel nun schon in deutsch ist.

deHarro, 2018-07-09:
Die Idee:
Ich habe eine manuelle [Steuerung mit zwei Joysticks ](http://www.harald-sattler.de/html/body_joystick-steuerung.htm#Joysticks) gebaut, die einfach parallel zu dem GRBL-Arduino auf die Stepper-Treiber wirkt. Natürlich ist eine gegenseitige Verriegelung eingebaut. Vereinfacht wurde das Unterfangen durch Verwendung der Interfaceplatine, die bei meinem Stepper-Set dabei war. Das Teil hat einen zweiten Eingang für eine manuelle Steuerung, den ich verwende.
Die beiden Joysticks habe ich zusammen mit zwei Schaltern (ein Taster, ein Schalter, beide haben eine steuerbare LED-Beleuchtung) in ein Alu-Gehäuse eingebaut, der zusätzliche Arduino sitzt im Rechner und 
bekommt seine Daten über drei geschirmte Leitungen. Ein Enable per Schalter ist ebenfalls dabei, ansonsten erzeugt der Arduino Step-Impulse, umso schneller, je weiter der Joystick ausgelenkt wird. 
Beschleunigungskurven konnte ich mir sparen, die mechanische Trägheit der Sticks reicht aus um die Motoren (sanft genug) anfahren zu lassen. 
Über die virtuelle serielle Schnittstelle des Arduino können die wesentlichen Parameter der Steperzeugung zur Laufzeit eingestellt werden.
Fehlt nur noch ein Interface zu GRBL-Plotter, so dass die über die Joystick implizierten Steps mitgerechnet werden... :-)
\Idee

Ok, du hast vorgeschlagen, den Joystick Jog-Commands ausgeben zu lassen, die der GRBL-Plotter dann an GRBL weiterleitet.
Ich werde mich dransetzen und mal checken, ob ich heute (im Sinne von "mittlerweile") in der Lage bin, den Code im Joystick soweit zu pimpen, dass ich alle drei Achsen gleichzeitig mit Jog-Commands füttern kann.
Um das zu testen bräuchte ich deine Unterstützung, denn auf einem Windows-Rechner ein Tool finden, das zwei virtuelle COM-Ports miteinander bidirektional (GRBL quittiert die JOG-Commands und gibt damit Grünes Licht für die nächsten Kommandos. Dieser Handshake war bei SCC nicht möglich, vielleicht der Grund für mein damaliges Scheitern?) koppelt, ist eine echte Herausforderung.
GRBL-Plotter müsste diese Funktionalität am besten bereits implementiert haben, damit ich zielgerichtet arbeiten kann.
Drawback dabei: Ich habe das schon mal mit SerialComCNC (SCC) versucht und bin gescheitert :(
Deine Entscheidung...

Harald

svenhb, 2018-07-09:
Mein erster Gedanke war die speziellen Kommandos - siehe
https://github.com/gnea/grbl/wiki/Grbl-v1.1-Jogging
Einfach durch zu reichen und ich gebe den grbl Status zurück.

deHarro, 2018-07-10:
Ja, ich denke, das sollte der Plan sein. Mit mehreren (virtuellen) COM-Ports kannst du ja umgehen (->zwei GRBLs) :-)
Bei Ulrich (SCC) bekam ich den Status nicht zurück und musste versuchen auszurechnen, ob GRBL schon wieder neue Kommandos verkraftet.
Meine damaligen Tests habe ich mit "Termite" gemacht, das ist ein Terminal-Programm, das das durchreichen zwischen zwei COM-Ports anbietet. Hat aber nicht so doll funktioniert und war dröge, weil ich immer zwischen den "Besitzern" des COM-Port wechseln musste (zw. programmieren und Runtime). Debug-Ausgaben gabs damit sowieso nicht.

svenhb, 2018-09-02:
Neu in Version 1.2.0.0
Neues serielles Interface, welches GCode Kommandos empfängt und an grbl weiterleitet.
Grbl-Stati (siehe grbl wiki) sowie Streamingstatus wird zurückgegeben

closed 2018-09-02 by svenhb

deHarro, 2018-09-02:
Hi Sven,
klingt gut :-)
Ich habe die Version geladen und gestartet, finde aber keine Option der seriellen SS für meinen Joystick.
Wo muss ich suchen?
Danke!
Harald

svenhb, 2018-09-03:
Im geschlossenen Issue zu schreiben ist keine so gute Idee...

Man findet die Steuerung über Menu- [Machine control - DIY control], bzw. [Maschine - Einstellungen] (vergessen umzubenennen) Oder wenn es bei Prog-start gleich erscheinen soll im Setup - Steuerung - DIY Steuerung.

svenhb, 2018-09-03:
Add some info to the wiki: https://github.com/svenhb/GRBL-Plotter/wiki/DIY-control