svenhb/GRBL-Plotter #23: Soft Limit

Gabse, 2018-08-17:
Would it be possible to add a keep in area on the machine coordinates, so that it gives an error when the machine is attempting to leave this area by jog or on the launch of a file if it would leave it during the program? The reason is so the user could set the maximal machines traveling coordinates and it would never accidently hit a hard limit or GRBL soft limit in normal operation. It would also be handy if this keep in area would be visible in the Preview window, so you could immediately see where the problem is.

Another thing: I was lately engraving a pretty big file (5MB) and had the issue that the CPU went up to 100%, and the stream started to paused for a few seconds after every movement. I soon discovered that it is probably the preview window, because when I switched to the full screen task manager (or any other window) the usage went down to 4% and the stream became smooth again. I updated to the latest version, and it is a lot better: only 90% CPU and a smooth stream, but I thought maybe there is a way to improve it even further. Fait to say that I am using an old Pentium4 notebook running XP, and the file was really huge.

Greetings Gabse

svenhb, 2018-08-17:
I was also thinking about showing machine / working-space dimension. I will implement.

Preview window: In older version I replaced the 'real' path by a picture during the run (to save CPU). Later on I implemented the zooming option wich was not compatible with the picture-view during the run, so I removed the picture replacement. 
I will implement an option to show either the real path (as current) or a picture to save CPU.

Gabse, 2018-08-17:
Wow, that went fast! 
Many Thanks 😀

svenhb, 2018-09-02:
Check new version 1.2.0.0
Option to show machine limits and alert, show tool position 
Option to reduce CPU load by showing background image during streaming

closed 2018-09-02 by svenhb