svenhb/GRBL-Plotter #27: Warning message with ambigous values when calling GRBl Settings ($$)

deHarro, 2018-09-03:
Hi Sven,
when I issue "$$" in the COM CNC dialog I get a warning message with some very big values:

[image]

If I calculate your given formula with the values of my GRBL I get 8.88886667 kHz what is way below the given maximum rating of 30 kHz for the step rate.

Please check the math behind the warning message.
Thanks! Harald

nota bene:
Issuing "Check GRBL" works _only once_ per start of GRBL-Plotter.
If you change some parameter and want to check whether the change is accepted, the values in the listbox  are not updated.
The warning mesage on the other hand reappears every time when clicking "Check GRBL".

deHarro, 2018-09-07:
The values for maximum frequency are correct now.

But what's about the "minimum feedrate"?
Where do you get information about GRBL minimum feedrate?
What value have I to change to get around this "Information" window?

And besides all: What does "Check RGBL" do?
Only a query for the machine settings?

svenhb, 2018-09-07:
>Where do you get information about GRBL minimum feedrate?
https://github.com/gnea/grbl/wiki/Known-Issues#doesnt-go-slower-than-x-mmmin-or-limitations-at-very-slow-step-rates

>And besides all: What does "Check RGBL" do?
Reading the grbl setting via $$ and giving information about min and max ratings (for arduino uno)

deHarro, 2018-09-07:
Hi Sven,
wir können hier noch ein paar mal hin und her "chatten", bis ich auf englsich rüber gebarcht habe, was ich eignetlich fragen wollte, aber ich versuche es mal auf  Deutsch:
Ich meinte, aus welchen Daten der GRBL-Parameter du errechnest, dass meine minimum feedrate zu klein sein könnte.
Ich verstehe diese Info-Box so, dass es da Handlungspotential gibt, ansonsten würde ich so eine Box weg lassen oder maximal einmal (während einer GRBL-Plotter-Sitzung) zur Info anzeigen.

Gruß, Harald

svenhb, 2018-09-07:
>Ich verstehe diese Info-Box so, dass es da Handlungsbedarf gibt
Den Handlungsbedarf kann ich ja nicht vorhersehen, bzw. teste die verwendeten Feedraten nicht darauf.
Da es aber in der grbl FAQ darauf hingewiesen wird, wollte ich die Problematik hier erwähnen: es gibt 'zu schnell' aber auch 'zu langsam'. Möglicherweise sollte ich diese Information anders darbieten?

deHarro, 2018-09-09:
> Den Handlungsbedarf kann ich ja nicht vorhersehen, bzw. teste die verwendeten Feedraten nicht darauf.

Zumindest die max Feedrates hast du aber berechnet und daher die Box ausgegeben (so hatte ich das jedenfalls verstanden, und du wohl auch, denn du hast ja dann die Werte neu richtig berechnet).

Deshalb nahm ich - mit den neu berechneten, korrekten und damit nicht mehr zu schnellen max Feedrates - an, dass mit den Untergrenzen der step rate ebenfalls etwas im Argen ist.

Und deswegen wollte ich wissen, auf welche Werte der GRBL Parameter sich die Warnung wegen zu langsamer Steprates bezieht.

Ok, zusammengefasst:
- Die Box kommt immer
- Die Box zeigt berechnete Werte für die max Steprate an
- Die Box zeigt _keine_ berechneten Werte für die min Steprate an
- Die Box teilt mit, dass es eine untere Grenze für die Steprate gibt, diese Grenze wird aber nicht notwendiger Weise von den eingegebenen GRBL Parametern verletzt, sondern nur informativ erwähnt

Stimmt das soweit?

deHarro, 2018-09-09:
Vorschlag für "andere Darbietung", falls ja:

Dann würde ich irgendwo vielleicht einen Knopf mit "?" drin anbieten, der bei Click darauf oder bei Hover darüber die Info-Box mit dem allgemeinen Text über die max und min Steprates öffnet.

Zusätzlich würde ich bei "Check GRBL" den oberen Teil (mit der max Steprate) aufblenden, wenn die eingegebenen Parameter die GRBL Vorgaben verletzen (wie das durch den Rechenfehler mit meinen Parametern der Fall war).

Der Teil über die minimale Steprate ist nach meiner Auffassung nur für den GUI-Entwickler (eines g-code Streamers) interessant, da der Anwender deines Programms an der min Steprate ja nichts ändern kann (stimmt das?)

svenhb, 2018-09-09:
Ich verstehe nicht "die Box kommt immer".
Meine Vorstellung war: normalerweise ist das COM-Fenster immer im Hintergrund und man nutzt es nur einmalig bei der Einrichtung (COM-Port, sowie grbl Setup über den Knopf unten "$$").
Mit "Check grbl" wird nur "$$" gesendet (um aktuelle EInstellungen zu erhalten) und die entsprechenden Berechnungen angezeigt.
Eine farbige Hervorhebung ist eine schöne Idee, aber die Umsetzung ist mir z.Zt zu aufwändig, da ich dann eine eigene Form basteln müsste. Momentan nutzte ich "MessageBox.Show".

Ich habe ja keinen Einfluss darauf wo der GCode herkommt, ob von GRBL-Plotter erstellt oder von einem anderen Programm und in GRBL-Plotter geladen. Evtl. sollte ich alle "F" Wörter auf den gültigen Bereich prüfen...

deHarro, 2018-09-09:
Ach so, ja, das habe ich nicht erklärt (oder nur mal kurz am Rande und auf Englisch).
"Die Box kommt immer" wenn ich im COM-Window auf "Check GRBL" klicke.
Und das habe ich halt mehrfach getan, weil ich ein paar Parameter über das Eingabefeld unten geändert habe.
Meine X- und Y-Achse fährt mit bis zu 2500 mm/min, die Z-Achse nur mit max. 600 mm/min, das musste ich unterschiedlich einstellen. Der für die Tests verwendete Arduino war jungfräulich bzw enthielt noch die Parameter von früher (alle Achsen 600 mm/min).

Außerdem habe ich das COM-Fenster sichtbar gehalten, weil ich ja für meine Tests mit dem Joystick die GRBL-Reaktion sehen wollte.
Das ist übersichtlicher als die Ausgaben im DIY-Fenster, das ja alle paar Sekunden mit irrelevanten Stati gefüllt wird.
Das zur Erklärung warum "immer...".

Und jetzt das von mir beobachtete und angemeckerte Phänomen:
Nach Start von GRBL-Plotter und klick auf "Check GRBL" kommt die Box und es werden _einmalig_ in der Listbox die Parameter aufgelistet.
Jetzt ändere ich etwas (Eingabe z.B. $110=2000 <Enter>) und will überprüfen, ob GRBL die Änderung akzeptiert hat. Also klicke ich wieder auf "Check GRBL", wieder kommt die Box mit den Warnung/Infos, aber der Listbox-Inhalt des CNC-Window ändert sich nicht, es werden also nicht nochmal die (jetzt geänderten) GRBL Parameter aufgelistet.

Gebe ich hingegen unten im Eingabefeld $$ <Enter> ein (oder klicke auf die $$-Taste), wird die Listbox aktualisiert und die Info-Box kommt nicht.

Ich würde hier erwarten, dass auch bei "Check GRBL" die Listbox aktualisiert wird.
Und wie gesagt, die Info-Box nicht jedesmal anzeigen, sondern nur, wenn wirklich der eine oder andere GRBL Parameter die GRBL Vorgaben verletzt.

deHarro, 2018-09-09:
> Eine farbige Hervorhebung ist eine schöne Idee, aber die Umsetzung ist mir z.Zt zu aufwändig,

Das meinte ich anders, ist von mir dumm geschrieben ,sorry.
Ich meinte, mit 
> Zusätzlich würde ich bei "Check GRBL" den oberen Teil (mit der max Steprate) aufblenden,

 Zusätzlich würde ich bei "Check GRBL" _den Text_ des oberen Teils in einer Box anzeigen.

deHarro, 2018-09-09:
Aber das kannst du natürlich alles genau so lassen, wie du es implementiert hast, ich wollte nur antworten auf deine Frage 
>  Möglicherweise sollte ich diese Information anders darbieten?

svenhb, 2018-09-09:
>es werden einmalig in der Listbox die Parameter aufgelistet.   
Ja das stimmt, werde ich ändern, solange dauert das Auslesen mit "$$" ja auch nicht. Eigentlich (nie wirklich getestet) sollte ein geänderter Wert in der gespeicherten Liste aktualisiert werden...

deHarro, 2018-09-09:
Soweit ich gesehen habe, wird die Liste nicht aktualisiert, der alte Wert bleibt angezeigt.

commit 55fa2cd1 (2018-11-05): Fixes #27
closed 2018-11-05 by svenhb