svenhb/GRBL-Plotter #29: 2. serial for Joystick, proposal: debug feature

deHarro, 2018-09-03:
Hi Sven,
I'm more or less sure that my Joystick issues commands to the serial port.
Only there is no visible reaction in GRBL-Plotter.
The stylus doesn't move, the coordinates do not change.

Can you please implement some sort of debug feature which allows the visualisation of the strings (or whatever is received on the serial port), perhaps easiest just into the listbox of the DIY control pad?
Would be fine. Thanks!
Harald

svenhb, 2018-09-03:
Hi Harald,
each received command will be shown in the window. I just used a 2nd arduino with grbl and received the known reset-startup message, which was then forwarded to grbl - which causes an error (because it was not a regular gcode).
Perhaps I need to offer settings for stop-bits? 
Or can you try to use grbl - com settings? (8 data bits, 1 stop-bit, no parity )

deHarro, 2018-09-04:
Hi Sven,
my Joystick uses "115200, 8, 1, n, no handshake", just the settings GRBL uses.

This is what I set on "Termite":

[image]

And this is the communication between joystick and GRBL:

[image]

COM14 ist the joystick, COM13 is GRBL (V1.1f).

Just got an idea... To keep GRBL satisfied when debugging with Termite I issue "?" commands on the joystick.
This perhaps garbles the communication between GRBL-Ploter and GRBL. 
I will stop the "?" in joystick  since those "?" are issued by GRBL-Plotter.
I will report...

Harald

svenhb, 2018-09-04:
Hello Harald,
in your screenshot I read "$J=G91X 1F1000" -> to be regular gcode, there must be no space between character and number: "X 1" is not a correct gcode - as I know.
To be on the save site I would use spaces between commands, like "$J=G91 X1 F1000"
Sven

svenhb, 2018-09-04:
You may should use "CR anhängen"

deHarro, 2018-09-04:
G-code with or without "CR" are treated similar.
I have different implementation (by chance, not deliberately :) for X- and Y-axis, respectively, and both are not shown in GRBL-Plotter.

deHarro, 2018-09-04:
In the meanwhile I discarded sending the "?" on my own in joystick, but no change in GRBL-Plotter.
Still no visible reaction in GRBL-Plotter.

svenhb, 2018-09-04:
Did you try just to connect another grbl - as I did? At least this should work...

deHarro, 2018-09-04:
Not yet... I will do and report

deHarro, 2018-09-04:
ok, nothing changed (despite the now omitted reset codes).
Again I get nothing until I reset the joystick arduino.

I try to append something (a "CR" or "LF" or both?) after the jog command, but that should not be neccesary following the jogging tutorial I linked above.

deHarro, 2018-09-04:
Ok, one step beyond... :-)

I appended "0x0D, 0x0A" (that is CR LF) and the commands appear in the list box of the DIYcontrolPad.

So what should happen next?
The coordinates in GRBL-Plotter do not change.

svenhb, 2018-09-04:
Very good! I'm happy too!

deHarro, 2018-09-04:
One more question: 
Do you mirror the "ok" responses of GRBL to the DIYControl port? (didn't examine that on my own yet)
That is essential for keeping the jog queue filled as much as possible.

deHarro, 2018-09-04:
If I encounter issues when dealing with the joystick port, should I raise a new issue?
(or report them here in this topic?)

svenhb, 2018-09-04:
All send messages will be shown in the DIY-control window - each grbl message in brackets "[ ]", but no 'ok'.
I would prefer to open a new issue and close this one - because communication is working now - right?

deHarro, 2018-09-04:
By the way...
How do you handle the immediate commands from the joystick?
When issuing a jog-abort command, then there is nothing like CR and/or LF, just the 0x83 on its own.
...
Ok, I see what you do (or better what you do not do).
Just as with my commands before the changes, the abort command is held in the input queue (of GRBL-Plotter, i assume) and is then forwarded to the listbox (and GRBL?) just in front of the next jog command (see the red circles):

[image]

To get jogging working as expected, you have to implement a means of handling those immediate commands as well.
Siede effect: If you get that done, there will be no need to finalize the jog commands with CR LF any more.

deHarro, 2018-09-04:
Hi Sven,
my question aims not on the listing, but on whether the "ok" is forwarded to my joystick arduino.

This is essential for me, when it comes to holding the jog queue stuffed.
When jogging continuously, it can happen, that the machine cannot keep up with the jog commands.
To prevent sending commands when the queue of GRBL is full, I have to stop sending commands.

Or is that problem handled by GRBL-Plotter for the joystick?

deHarro, 2018-09-09:
Hi again!
You stated elsewhere (see https://github.com/svenhb/GRBL-Plotter/issues/34#issuecomment-419643553), I should recon how fast I can send additional jog commands. That would be possible but complicated and time consuming for the Arduino, I fear.

On the other hand they mention, one has to wait for the "ok" from GRBL before the next jog command is issued:

>Basic Implementation Overview:
    - Create a loop to read the joystick signal and translate it to a desired jog motion vector.
    - Send Grbl a very short G91 incremental distance jog command with a feed rate based on the joystick throw.
    - **Wait for an 'ok' acknowledgement before restarting the loop**.
    - Continually read the joystick input and send Grbl short jog motions to keep Grbl's planner buffer full.

I think at the moment you do _not_ forward the "ok" from GRBL to the DIY port.

svenhb, 2018-09-09:
I don't want to pass through the 'ok' - it's information overkill in my eyes...
Check https://github.com/svenhb/GRBL-Plotter/blob/master/GRBL-Plotter_1208_test.zip

deHarro, 2018-09-10:
Ok, please close this issue then.
Thanks anyway so far!! :)

commit 72267c3b (2018-11-05): Fixes #29, #33, #34
closed 2018-11-05 by svenhb