svenhb/GRBL-Plotter #30: Tooltip of cyan "Abmessungen" field is nearly half a meter in size

deHarro, 2018-09-03:
Hi Sven,
sorry for the many issues I stumble over, hope it's ok to report...

If I hover the mouse over the cyan filed named "Abmessungen" a tooltip appears which reaches the right border of my monitor.

[image]

I think the numbers are not very meaningful.
At least they should be shorter ;-) to get the sense.
Harald

svenhb, 2018-09-03:
What did you do after starting the program? The "Abmessungen" box should stay white as long as no data is loaded...

deHarro, 2018-09-04:
This is what I get, when I start GRBL-Plotter:

[image]

No data is loaded (by me), GRBL is online on COM13.