svenhb/GRBL-Plotter #31: Virtual Joystick control - Jogging doesn't stop when releasing the mouse

deHarro, 2018-09-03:
HI Sven,
when you use the virtual Joystick control, you should send an "jog abort" command immediately after releasing the mouse (when the "cursor" returns to the middle.
You omit this commmand and GRBL moves until all jog commands in its queue are through.
This may result in overshoot.

The jog abort clears the queue and cleanly decelerates the mill as fast as possible to a stop.

I think this behavior relies on the speed and acceleration values GRBL knows about the machine and the values you entered as defaults for the joystick settings.
Harald

svenhb, 2018-09-03:
I will implement "stop jogging" when releasing the mouse button.

deHarro, 2018-09-07:
The virtual joystick now issues 0x85 when releasing the mouse -> close

closed 2018-09-07 by deHarro