svenhb/GRBL-Plotter #33: DIYcontrol port, "incorrect" messages block the input buffer

deHarro, 2018-09-04:
When working with the DIYcontrol port we deduced (see issue #29), that the jog commands have to be finalized with CR LF to be recognized by GRBL-Plotter

If any character (or string or whatever) reaches the port, this message is not processed until the algorithm sees a CR LF.

With short commands this seems to work well enough (one character abort jog commands are ignored but are "processed" (at least accepted and ignored) in front of the next finalized command).
But if longer inputs occur (e.g. _some_ not finalized jog commands), this stuffes the input queue of GRBL-Plotter and further commands are not processed.

svenhb, 2018-09-06:
Normal commands must be finished with CR LF, but real time commands not.
See:  https://github.com/gnea/grbl/wiki/Grbl-v1.1-Interface#grbl-interface-basics
"The primary way to talk to Grbl is performed by sending it a string of characters, followed by a carriage return."

deHarro, 2018-09-06:
You are right, I just wanted to give feedback which helps you making the interface more robust.
I stumbeled over this behavior since my joystick previously sent commands without CR LF.
This should not occur anymore.

But again:
Blocking of the input queue because of invalid entries should be handled in some way. 
Just my 5 cent...

Harald

commit 5be829b8 (2018-11-05): Fixes #33
closed 2018-11-05 by svenhb