svenhb/GRBL-Plotter #34: DIYcontrol port, immediate commands are not supported

deHarro, 2018-09-04:
To support jogging with dedicated joystick hardware according to "GRBL V1.1 Jogging" via this second DIYcontrol port, the handling of immediate commands has to be supported.

This implies:
- The receiving thread in GRBL-Plotter has to accept single characters over the serial port
- GRBL-Plotter _must not_ wait for CR LF to treat the received character
- Instead, GRBL-Plotter has to accept every single character on this port when it arrives and forward it immediately to GRBL on the CNC port.

Besides that, to prevent accidental forwarding of jog commands, GRBL-Plotter has to block input on the DIYcontrol port as long as a CNC job is running.
I suppose, that is already guaranteed since there is the "virtual joystick" in GRBL-Plotter.

svenhb, 2018-09-04:
I will implement somthing like this:
receiving char by char, filtering char > 0x7F -> send real time command, or collect further chars until /r/n is reached, then send by regular function.

deHarro, 2018-09-04:
Might work, lets see how it performs... :-)
Thanks!

svenhb, 2018-09-05:
Try test version https://github.com/svenhb/GRBL-Plotter/blob/master/GRBL-Plotter_1202_test.zip
were I implemented my suggestion. Could not test real time commands, because no hardware which can send, I hope it works.

deHarro, 2018-09-06:
Something went wrong :(
You double every Jog command and GRBL doesn't understand that.

[image]

[image]

Coordinates in GRBL-Plotter dont change.

It's not clear, whether the realtime command is forwarded to GRBL or not.

(You said, you have a spare Arduino. Are you willing to buy a [joystick ](https://www.ebay.de/itm/JoyStick-Breakout-Module-PS2-Joystick-Game-Controller-for-Arduino/201380684238)at ebay?
If so, I happily send you the sources of my joystick controller for testing.
But we can try debugging together, no problem)

deHarro, 2018-09-06:
This is my test equipment at the development computer:
[image]

[image]

The right Arduino has one stepper on the X axis to get the feeling when stepping, since one is fooled by the changing numbers in the different applications I use when developing, sometimes.

The left Arduino runs the joystick firmware, I can control X and Y, Z is fixed centered in software.
If necessary I can attach one button, the thumb joystick features a NO switch by pressing the knob.

svenhb, 2018-09-06:
But realtime command works?

deHarro, 2018-09-06:
As I wrote, I cannot verify that.
To proofe the functionality, GRBL-Plotter and GRBL have to execute the jog commands and _then_ the jog-cancel can cut in and stop the motion underway.

As long as the jog commands are not forwarded correctly... no chance to test. Sorry.

svenhb, 2018-09-06:
Try new version, hope it works as expected: https://github.com/svenhb/GRBL-Plotter/blob/master/GRBL-Plotter_1203_test.zip
Perhaps have a look to issue #27, #29, #30, #31, #33,

deHarro, 2018-09-06:
Sorry, same effect as before:

[image]

svenhb, 2018-09-06:
Aber jetzt: https://github.com/svenhb/GRBL-Plotter/blob/master/GRBL-Plotter_1204_test.zip

deHarro, 2018-09-07:
Ok, one step beyond :+1: :-)
The jog commands are forwarded perfectly now.

But how sure are you, that 0x85 chars are transmitted asynchronously and bypassing the already sampled strings (or string parts)?

As far as I can judge from my seeings, this part of the job has potential... will say: That doesn't perform well.

I just fed some seconds of very small and slow but following sequently fast movements  into GRBL-Plotter, the stepper motor follows.
Fine so far.  
(Explanation: "very small": X=1 mm, "slow": F=84, "following fast": every 10 ms)

Then I released the joystick, what should issue the jog-abort commmand (0x85). 
But the stepper happily moved on for around half a minute or so, before the somewhat faster movement came to GRBL.
I even changed rotation direction for some time. This reached my motor after around a minute or so.

This leads me to assumption that the abort command is **not** forwarded passing by the already sampled jog commands.

I try to manage to visualize the characters from my joystick to GRBL-Plotter on the fly...

deHarro, 2018-09-07:
Ok, got a serial port spy up and running...
I send consecutive jog commands and when releasing the knob of the joystick, one or two 0x85 are issued:

[image]

So far I'm confident, that the culprit lies somewhere in GRBL-Plotter.

Perhaps you managed to deal all well on the listening port from the joystick, but forgot to handle the bypassing on the sending port to GRBL?

svenhb, 2018-09-07:
I need to check this evening...
>But how sure are you, that 0x85 chars are transmitted asynchronously and bypassing the already ampled strings (or string parts)?
At least the 0x85 are shown in the DIY-control window, with explanation text?

deHarro, 2018-09-07:
Hi Sven,
is "At least the 0x85 are shown in the DIY-control window, with explanation text?" in the comment above a question from you to me?

If so, no, I do not see any "0x85 with explanation" in the DIY window.

Here is what I get:

[image]

This is directly after closing/opening both ports (to clear the windows) and performing a short positive X move with the joystick.

Now again a short move in positive X:

[image]

You see, there is a "?" in front of the first jog command of this second move.
And you see, this "?" is echoed to GRBL _in front of the next jog command_ to GRBL, too! (You can count the jog commands in both pictures to verify this)

svenhb, 2018-09-07:
I mean your post 2hours ago: "I send consecutive jog commands and when releasing the knob of the joystick, one or two 0x85 are issued"
When you send 0x85 do see it in the DIY-control window?

deHarro, 2018-09-07:
No, I see only my jog commands and those "?" characters.

svenhb, 2018-09-07:
The '?' characters matches with your Jog-Stop command?

deHarro, 2018-09-07:
I would say yes.

svenhb, 2018-09-07:
New try: https://github.com/svenhb/GRBL-Plotter/blob/master/GRBL-Plotter_1205_test.zip

deHarro, 2018-09-07:
Hi Sven,
no "?" any more in the in-/output, you now issue a "0x3F Status Report Query" in both log windows.

Hopefully this leads us to a solution, but just now the jog commands are still not cancelled.

0x3F is not what I send, I send 0x85 (actually I use "Serial.write(133);" in my code).

So, where does this 0x3F Status Report Query come from?

svenhb, 2018-09-07:
Try to send another realtime command below 0x85, like '!' or '~'. What happens then? Perhaps you need to enable your 7th bit to send? (I know doesn't match with 0x85 <-> 0x3F)

svenhb, 2018-09-07:
Possible real time comands I forward as real time command, (in addition to >0x7F)  are 0x18, '?', '~', '!'

deHarro, 2018-09-07:
Ok, I will try first with "!"

svenhb, 2018-09-07:
ok, now I convert unicode to byte - hope is now reading correct values: https://github.com/svenhb/GRBL-Plotter/blob/master/GRBL-Plotter_1206_test.zip

deHarro, 2018-09-07:
I'm sorry, still 0x3F

[image]

deHarro, 2018-09-07:
Hi Sven,
did you encounter the following source:
[SerialPort Encoding](https://blogs.msdn.microsoft.com/bclteam/2006/05/26/serialport-encoding-ryan-byington/)

In short: "You have two options; one would be to use the **API’s that deal with byte[] and bypass the Encoding**, or to use the correct Encoding. However finding the correct encoding is a bit tricky..."

svenhb, 2018-09-07:
No, i didn't know, I will check.
I started with the function "serialPort.ReadTo("\r\n");" which is used in the other serial-windows. But with this I can't filter the real time command. Then I switched to "serialPort.ReadExisting()" (returning a string) but, as we figured out, chars above 0x7F are a problem. 
Next try is "SerialPort.Read" which fills a byte-array. I hope this solves the problem (tomorrow).

svenhb, 2018-09-08:
Tried "SerialPort.Read": https://github.com/svenhb/GRBL-Plotter/blob/master/GRBL-Plotter_1207_test.zip
Hope is fixed now...

deHarro, 2018-09-08:
Hi Sven,
I think, we are nearly through :-) 

[image]

Your port now recognizes characters greater than 0x7F and forwards them to GRBL (as far as I interprete the output of both listboxes).

Only GRBL doesn't stop immediately but continues to move til the end of the issued jog commands.

So I think, now you have to look at the sending side aimed at GRBL.
Perhaps you have to flush (or delete) the output queue instantly after fordwarding the 0x85 to GRBL?

svenhb, 2018-09-08:
Too bad... I found also another problem: jogging during pause of a stream doesn't work anymore...
You could follow this, to avoid sending too much jog-commands: https://github.com/gnea/grbl/wiki/Grbl-v1.1-Jogging#how-to-compute-incremental-distances

svenhb, 2018-09-09:
Problem solved, 0x85 will also clear my internal buffer
https://github.com/svenhb/GRBL-Plotter/blob/master/GRBL-Plotter_1208_test.zip

deHarro, 2018-09-10:
Hi Sven,
seems to work flawlessly, very fine!

I'm on implementing the math for step size, speed and proper command frequency according the joystick throw. Hope the AtMega is fast enough...

closed 2018-11-05 by svenhb

papillion47, 2024-10-09:
Ben acemi bir CNC ve GRBL kullanıcısıyım. İlk başta bana önerilen linkten GRBL yi yükledim. Ve wiki sayfalarından öğrenerek tüm ayarları doğru şekilde yaptım. Vectric Aspire de hazırladığım tap uzantılı işlerimi gayet sağlıklı olarak birkaç yıldır yapmaktaydım. Geçenlerde ne olduysa visualizer ve heightmap çalışmamaya, ekranda hiçbir şey göstermemeye başladı. GRBL yi defalarca silip yeniden yüklemek fayda etmedi. İnternetten bularak GRBL benzeri olan Candle programı yükledim. o zaman da visualizer çalıştı fakat bu defa manuel kumanda ettiren jog komutları aktif olmadı. O komutları kullanmadan gereken spindle yeri ayarları yapılamıyor. Bu konuda bir yardımı olacak çıkarsa çok mutlu olacağım. Aşağıda son ayarlanan durum ekranımı veriyorum. Teşekkürlerimle..
[image]

svenhb, 2024-10-09:
What do you get if you enable "Show real time status report", before connecting the controller?
Should look like this:
[image]

papillion47, 2024-10-10:
Benim grbl de böyle bir seçenek yok.

svenhb, 2024-10-10:
You should use latest release of GRBL-Plotter: https://github.com/svenhb/GRBL-Plotter/releases

commit 5064d41b (2018-11-05): Fixes #34