svenhb/GRBL-Plotter #35: Start from select line code

78Matteo78, 2018-09-04:
Hello, thanks for the great job.
I wanted to ask if it is possible to start my cnc from a selected line of code. In order to complete a long work in several moments.
Best regards

78Matteo78, 2018-09-04:
> How to handle commands before the selected line? I need to run the gcode from the beginning to get all mode changing commands (check  http://linuxcnc.org/docs/html/gcode.html ) but exclude real movements...
> I have to think about that..

For example, with mack 3 you can select an intermediate line of the gcode program and start from that point. This allows me to complete a long project in several phases. But maybe it is gbrl that does not allow this function. Thanks anyway
--
Inviato da Libero Mail per Android martedì,

svenhb, 2018-09-12:
After stopping the job in the middle, do you just leave the program? Or doing another job?
How would you know which line to continue?

svenhb, 2019-01-03:
In next version I implemented following work-flow (not fully tested, not ok for tool change or subroutines):
If you want to restart a job later, 
- start job, current GCode will be saved as "lastProcessed.nc"
- press Play/Pause, wait until button flashes yellow - current parser state will be saved as "lastProcessed.xml"
- some time later - be sure machine coordintes are still ok - load "lastProcessed.nc". You will be asked if you want to continue old job. If yes, parser state will be restored - press Play to continue the job

78Matteo78, 2019-01-03:
Ok Thanks a lot of good work
--
Inviato da Libero Mail per Android giovedì,

svenhb, 2019-01-04:
Implemented in Version 1.2.3.x please check for functionality

closed 2019-01-04 by svenhb

78Matteo78, 2019-01-04:
I have tested the program and it seems like a great job. Thank you so much venerdì,

78Matteo78, 2019-11-07:
Hi svenhb
It is possible to have in gbrl plotter a simulator that can execute the tool path even if not connected to Arduino, and to indicate the total execution time of the project. Thanks again for the fantastic project

svenhb, 2019-11-08:
I will see if I can get some code to implement.
Just moving the cursor through the code and watching the marker position inside the 2D-view is probably not enough?

78Matteo78, 2019-11-08:
Now to simulate the tool path with a player, but more importantly, to see the expected code execution time, I use the software (NC CORRECTOR V4.0).

Once the gcode is loaded, it provides you with various work data.
More than the simulation player I would like to know the expected execution time.
Thanks again

svenhb, 2019-11-16:
Made a new release:
* Add rough process time estimation - shown in "GCode-Dimension"
Hope this is what you were asking for...

78Matteo78, 2019-12-05:
Hi svenhb
I noticed that in the last versions of the program the coordinates of the tool at startup are reset to zero.
This does not allow to use the function (last process) as in the previous version

svenhb, 2019-12-05:
Check [Setup - Flow Control - "Restore work coordinates..."]
I think this is missing?

78Matteo78, 2019-12-05:
Thanks svenhb
I'm sorry

78Matteo78, 2022-04-19:
--
Inviato da Libero Mail per Android