svenhb/GRBL-Plotter #36: Get the height information of a work piece by a camera

wwwlxs, 2018-10-19:
Hi Svenhb ，Can get the height information of a work piece by an USB camera’s focal distance information  to getting height map data instead of the probe for faster processing？

wwwlxs, 2018-10-19:
or use kinect/intel realsense dephth camera

svenhb, 2018-10-20:
Finally you want an external height measurement to do offset of z axis?
-How to readout camera’s focal distance information ? Sounds not very accurate.
-Connecting a kinect sounds like overkill
Perhaps you build an external device an connect it via "DIY control" (https://github.com/svenhb/GRBL-Plotter/wiki/DIY-control), Then  I need to implement some new commands to process the information.

wwwlxs, 2018-11-02:
Hi Svenhb ，Can get the height information from panasonic HG-C series measurement sensor? As compact as a photoelectric sensor, the CMOS laser sensor allows to display actual measured distance as accurately as a displacement sensor.
https://www.panasonic-electric-works.com/se/hg-c-measurement-sensor.htm

svenhb, 2018-11-02:
It may work, but I don't get the desired application:
Usually you want to know the distance between tip of tool and workpiece at an given Z-pos. to compensate tool-offset (each tool shows different length when fixed in spindle).
Then your sensor needs to measure bottom up the tip of the tool - does it work?
Then you need to read the distance and put in manually into software. Or how to get the distance-information into GRBL-Plotter? (perhaps via 'DIY-Control').

Or shows the tool always the same offset and you just need to know the Z-pos of the workpiece surface?
To adapt any kind of sensor, you need to build the hardware, which can be connected via COMx to the PC, sending the height information in an known format.
Then I can make an interface to read and process the information.

wwwlxs, 2018-11-02:
or CX-441 | Compact Photoelectric Sensor https://www3.panasonic.biz/ac/e/search_num/index.jsp?c=detail&part_no=CX-441

wwwlxs, 2018-11-02:
If it can use the Panasonic HG-C, it may be faster proceesing. If the Panasonic cx441 can be used, it can obtain data without touching the paper, thus preventing the paper from being deformed by the probe.

wwwlxs, 2018-11-10:
USB Data Acquisition Module
[image]
[image]
[image]

wwwlxs, 2018-11-10:
USB Data Acquisition Module
[USB Data Acquisition Module mp4.zip](https://github.com/svenhb/GRBL-Plotter/files/2568451/USB.Data.Acquisition.Module.mp4.zip)

svenhb, 2018-11-10:
I think there are two options:
1. a lot of work for me to get and process the external height data. Because it's not compatible with the current implementation - see below.
2. you write your own software, perhaps using the 'DIY-Control' interface. You then send xy coordinates to GRBL-Plotter, wait for idle, then check and save you height information in the format you like...

The current implementaion is: I move the specific XY position and start probing, the probing answer from grbl-hardware ([PRB:0.000,0.000,0.000:0]) triggers my heigth-measurement module to store the height information.

wwwlxs, 2018-11-10:
Thanks for your help！

wwwlxs, 2018-12-31:
happy new year!svenhb
now i use another Arduino to send heigth-measurement (Z position with "mm") to "diy-control"interface real time,
could you help me to take the heigth-measurement from"diy-control"to heigth-measurement module to store the height information instead of the probing answer from G38.3?

svenhb, 2018-12-31:
Happyend New year for you too :-)
We still need to wait one hour. 
You should receive the actual Position each 1/10 second. Perhaps I need to extend this Interface. 
Give me some ideas please. ..
Best regards 
Sven 
-- 
Diese Nachricht wurde von meinem Android Mobiltelefon mit GMX Mail gesendet.

"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/svenhb/GRBL-Plotter/issues/36#issuecomment-450693341",
"url": "https://github.com/svenhb/GRBL-Plotter/issues/36#issuecomment-450693341",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

wwwlxs, 2018-12-31:
thanks
i conect panasonic HG C1050 sensor to Arduino UNO, than send the actual Position to 'DIY-Control' with this code:

void setup() {
  Serial.begin(115200);
}

void loop() {
 float distance;
distance=map(analogRead(0),0,1023,-1500,1500); 
Serial.println(distance/100);
delay(100);
}

wwwlxs, 2018-12-31:
and the DIY-control display height information like this:
-8.41
-8.41
-8.38
-8.41
-8.41
-8.38
-8.41
-8.38
-8.38
-8.38
-8.41
-8.38
-8.41
-8.38
-8.38
-8.41
-8.41
-8.41
-8.38
-8.41
-8.41
-8.41
-8.38
-8.38
-8.41
-8.41
-8.38
-8.44
-8.38
-8.38
-8.38
-8.41
-8.41
-8.38
-8.38
-8.38
-8.38
-8.38
-8.38
-8.41
-8.38
-8.38
-8.38
-8.41
-8.38
-8.41
-8.38
-8.38
-8.38
-8.44
-8.38
-8.38
-8.41
-8.38
-8.41
-8.38
-8.38
-8.38
-8.38
-8.38
-8.38
-8.41
-8.38

wwwlxs, 2018-12-31:
and the DIY-control display height information like this:

Open COM18
< -8.82 
< -8.82 
< -8.85 
< -8.91 
< -8.85 
< -8.85 
< -8.85 
< -8.85 
< -8.82

svenhb, 2018-12-31:
Ok, what should I Do with the heißt Information? Should I Set the actual Z to this value? 
-- 
Diese Nachricht wurde von meinem Android Mobiltelefon mit GMX Mail gesendet.

svenhb, 2019-01-01:
If a trigger is needed, a custom button could be programmed with a specific command, hidden in a GCode-comment, perhaps "($TRG1)" which could then lead to a message via the DIY control like "[TRG1]".

In a next version I will support this: Any single string from a custom button like "($abc)" will be sent via DIY-Control as "[abc]"

wwwlxs, 2019-01-01:
Thank you very much！I think there are 2 designs to avoid moving the z-axis, which saves a lot of time: The first is to increase the way to get the z-position information in the grbplotter's height-map module. The position of the previous z is g38.x feedback, and now it is changed to the instantaneous z value received directly from the diy-control. This z value (in millimeters) is sent by the second Arduino through another COM. 
The second is Directly modify the motion module of g32.1-38.4 of grbl firmware, and replace their motion mode with "get the z value of the second arduino"

wwwlxs, 2019-01-01:
Maybe send "G92 Z-8.82 " to set Z value?
to make the heightmap like this:

<?xml version="1.0" encoding="utf-8"?>
<heightmap MinX="0" MinY="0" MaxX="25" MaxY="25" SizeX="185" SizeY="185">
  <point X="0" Y="0">-8.82</point>
  <point X="1" Y="0">-8.82</point>
  <point X="2" Y="0">-8.85</point>
 ...

wwwlxs, 2019-01-01:
Maybe send "G92 Z-8.82 " to set Z value?
to make the heightmap like this:
`<?xml version="1.0" encoding="utf-8"?>
<heightmap MinX="0" MinY="0" MaxX="25" MaxY="25" SizeX="85" SizeY="85">
  <point X="0" Y="0">-8.82</point>
  <point X="0" Y="1">-8.82</point>
  <point X="0" Y="2">-8.85</point>
  <point X="0" Y="3">-8.91</point>
  <point X="0" Y="4">-8.85</point>
 .............`

svenhb, 2019-01-01:
Ahh, you want to genrerate a height-map from your z-values?
Then I should modify the code for surface-scan to ask for the alternative Z-value instead of performing G38.3 ...
You may send frequently the actual height as "Z-8.82" to the DIY-Control and I will take over this value for the height map...

wwwlxs, 2019-01-02:
> Ahh, you want to genrerate a height-map from your z-values?
> Then I should modify the code for surface-scan to ask for the alternative Z-value instead of performing G38.3 ...
> You may send frequently [...]
Yes， I want to genrerate a height-map from  z-values by sensor,the second Arduino send hundreds of  the actual height per second to Diy-control,so it just only moving XY-axis and avoid moving the z-axis by G38.3 to get the actual height.This will increase the speed by  hundreds of times and save a lot of time to creat height map。

wwwlxs, 2019-01-02:
Yes， I want to genrerate a height-map from z-values by sensor,the second Arduino send hundreds of the actual height per second to Diy-control,so it just only moving XY-axis and avoid moving the z-axis by G38.3 to get the actual height.This will increase the speed by hundreds of times and save a lot of time to creat height map。

svenhb, 2019-01-02:
I found following solution (in next version): you send your values in the format "(PRB:Z-5.12)" - in brackets to be processed as comment by grbl (if no streaming, this data is passed through the arduino and would move the z axis).
If GRBL-Plotter received a valid message, the original probing z-value will be replaced by the new one.
To speed up probing you set "Max-depth" to 0.01 and "Save height" to 0. So I don't need to change my workflow with probing.

wwwlxs, 2019-01-02:
If I set "Max-depth" to 0.01 and "Save height" to 0, the z-axis will move 0.01mm, Does the machine shake and affect accuracy?
 OK，thanks svenhb！I am looking forward to the arrival of the next version.

svenhb, 2019-01-02:
Good hint...I made a work-arround to avoid G38.
Set Max-depth to 0, then an internal flag will be set.
After reaching new position (status switches to IDLE) the Z value from DIY-Control will be read, so synchronization is possible (check for <IDLE, send Z value)

svenhb, 2019-01-04:
Implemented in Version 1.2.3.x, please check functionality

closed 2019-01-04 by svenhb

wwwlxs, 2019-01-04:
thanks!
when ,open diy-control,"COM CNC' windows and "DIY-control"window are blinking all the time,and can not open the"creat height map'

wwwlxs, 2019-01-04:
it sucessfull！thanks！thanks svenhb！

svenhb, 2019-01-04:
>when ,open diy-control,"COM CNC' windows and "DIY-control"window are blinking all the time,and can not open the"creat height map'

Is it still a problem? I need more information, you may have screenshots?

wwwlxs, 2019-01-04:
i upload the screenshots to github but it fail，i can upload the video an screenshot to my instagram:instagram.com/wwwlxs

wwwlxs, 2019-01-04:
it sucess now！faster than ever! i can send the screenshot to you email.

svenhb, 2019-01-04:
I registered to instagram right now... But mail is also ok

svenhb, 2019-01-04:
Thanks for sharing. Why text font is different to my settings? Text is too long...
Is it default on your PC or did you change something?
[image]

wwwlxs, 2019-01-04:
it default on my pc.my pc is macbook pro2009 with 1280x800pix,the UI and font on windows 7 can not display well

svenhb, 2019-01-04:
Most text is written in the font "Microsoft Sans Serif". Perhaps you need to install this font?

wwwlxs, 2019-01-05:
If the software can arrange the windows like the Photoshop UI, using a scalable windows, it should be fine. It would be better if there is a Mac OS version.
 In addition, ‘scan-heightmap’ will pause every point to read the Z-axis , and then read the information of the next point. If it does not stop, the speed will be faster.
Sometimes such a message pops up:‘offset and lenght were out of bounds for the array or count is greater than the numbers of element from index to the end of the source collection’，Probably because of ‘delay(<200)’.

svenhb, 2019-01-05:
>It would be better if there is a Mac OS version.    

Not from me - sorry
>If it does not stop, the speed will be faster.

Also not possible - then you need a complete different algorithmus:   
Move from line-beginning to line-end, in certain time-steps take a sample of the Z-value. Most probably time steps are still to long (isn't there a limit in windows of 20ms?).
Perhaps you can speed-up acceleration $120 and speed $110 ?  
>Sometimes such a message pops up:‘offset and lenght were out of bounds for the array or count is greater than the numbers of element from index to the end of the source collection’，Probably because of ‘delay(<200)’.

What is your step-size in x and y or amount of data-points?  
 What do you mean with "Probably because of ‘delay(<200)’."?

wwwlxs, 2019-01-09:
“Most text is written in the font "Microsoft Sans Serif". Perhaps you need to install this font?”
----yes ，I have installed "Microsoft Sans Seif",and it still shows the same effect

"Most probably time steps are still to long (isn't there a limit in windows of 20ms?).
Perhaps you can speed-up acceleration $120 and speed $110 ?"
---setting:$110=2475and $120=200,$121=500

"What is your step-size in x and y or amount of data-points?
----Do you mean $110 setting? I setting $110=100,$0=10,$1=25

What do you mean with "Probably because of ‘delay(<200)’."?"
-----The sencond Arduino print" (PRB Z-3.33)" to DIY-control then delay 200mS per time,it no problem. but if i setting "delay 10mS", a message pops up:‘offset and lenght were out of bounds for the array or count is greater than the numbers of element from index to the end of the source collection’

wwwlxs, 2019-01-09:
Another issues:In "Setup-Graphics Import-G-code generation: Pen up/down traslation",When I chose "Z-Axis:Use Z-Axis and spindle" ,the"Z-Height save,"and"Z-height engrave" in G-code have nothings change when I "Apply Height_Map" from Surface scan.

E.g: 
Z-Height save:10;  
Z-height engrave:-5;

(without Height map) G-code:

"( Text import by GRBL-Plotter )
( G-Code lines: 156 )
( Pen Down/Up : 17 times )
( Path length : 0.0 units )
( Duration ca.: 0.6 min. )
( Conv. time  : 00:00:00.0026323 )
G54 (Setup - GCode-Header)
( Text: 123 )
( Font 'Serif bold italic' not found )
( Using alternative font )
M05
G00 Z10.000 
( Char: 1)
G00 X4.571 Y6.476  
G01 Z-5.000 F1000 
M3 S1000
G01 X2.667 Y0.000 F2000 
G01 X3.429 Y0.000  
M05
G00 Z10.000 
G00 X5.714 Y8.000  
G01 Z-5.000 F1000 
...
"

when I apply Height-Map, g-code:

"( Text import by GRBL-Plotter )
( G-Code lines: 156 )
( Pen Down/Up : 17 times )
( Path length : 0.0 units )
( Duration ca.: 0.6 min. )
( Conv. time  : 00:00:00.0026323 )
G54 (Setup - GCode-Header)
( Text: 123 )
( Font 'Serif bold italic' not found )
( Using alternative font )
M05
G00 Z10.000
( Char: 1)
G00 X4.571 Y6.476 Z10.683
G01 Z-5.000 F1000
M3 S1000
G01 X3.619 Y3.238 Z-4.068
G01 X2.667 Y0.000 Z-3.823
G01 X3.429 Y0.000 Z-3.858
M05
G00 Z10.000
G00 X5.714 Y8.000 Z10.524
G01 Z-5.000 F1000
M3 S1000
G01 X4.952 Y6.476 Z-4.342
G01 X4.000 Y3.238 Z-4.089
G01 X3.048 Y0.000 Z-3.840
M05
...
"

"G00 Z10.000 "and "G01 Z-5.000 F1000" change nothing,So when the Pen down,it will move deeper than paper surface .

svenhb, 2019-01-09:
I fixed it: https://github.com/svenhb/GRBL-Plotter/blob/master/GRBL-Plotter_1232_publish.zip

>"What is your step-size in x and y or amount of data-points?  

I mean the raster width for surface scan.
>What do you mean with "Probably because of ‘delay(<200)’."?"

Did you try values between 100 and 200?
Perhaps problem is my grbl refresh rate of 200ms (sending ?, receiving IDLE information)

svenhb, 2019-01-12:
Perhaps you can speed up probing, using my first idea:
>I found following solution (in next version): you send your values in the format "(PRB:Z-5.12)" - in brackets to be processed as comment by grbl (if no streaming, this data is passed through the arduino and would move the z axis).
If GRBL-Plotter received a valid message, the original probing z-value will be replaced by the new one.
To speed up probing you set "Max-depth" to 0.01 and "Save height" to 0. So I don't need to change my workflow with probing.

Because after probing response from grbl comes immediately, 
compared to the second version where we need to wait for IDLE response from ask of real time status report.

A quick test without hardware shows 14 seconds for 1st version (max depth = -0.01)
and 24 seconds for 2nd version (max depth = 0)

wwwlxs, 2019-01-12:
ok，thank svenhb，How to send the values to be processed as comment by grbl using 1st version？

svenhb, 2019-01-12:
max depth =<0 (-0.001) -> verion 1
max depth = 0 -> version 2

wwwlxs, 2019-01-12:
I mean my Z values sending to DIY-Control also in the format "(PRB:Z-5.12)" ？and set max depth to -0.001 in Surface scan windows？

wwwlxs, 2019-01-12:
yes!thanks！ The 1st version faster than ever！ but max depth can be set to-0.01,but can not be set to-0.001.
I took a video on instagram

wwwlxs, 2019-01-12:
Another iussus when apply height map:  ,When Z value from height-Map bigger than Gcode Z value (Pen up /down by G00 ) ,the pen down then up ,and writting in the air.

wwwlxs, 2019-01-12:
[gcode and heightmap.zip](https://github.com/svenhb/GRBL-Plotter/files/2752435/gcode.and.heightmap.zip)

svenhb, 2019-01-12:
>Another iussus 

ok thanks, I will check this soon.

wwwlxs, 2019-01-12:
Thanks，Plotter suddenly stopped during the scanning height-map.
COM window：

`[PRB:157.130,227.800,-12.875:0]
[PRB:157.030,227.800,-12.875:0]
[PRB:156.930,227.800,-12.875:0]
<
< !!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
< error:24 	More than one g-code command that requires axis words found in block."

< ok
[PRB:156.730,227.800,-12.875:0]
< ok
< ok
< ok`

svenhb, 2019-01-12:
>Another iussus 

Your Z goes positive, because height map has positive Z...
Workarround: you press "set max value = 0.0" in surface scan window before applying map...
Should I adapt G0 Z-values also in gcode?

wwwlxs, 2019-01-15:
If I input the z-axis position information through grbl's Arduino uno A4 IO input, can grblplotter read A4 IO？so that I can avoid using the second Arduino through the second COM, will it be more stable?

svenhb, 2019-01-15:
>fast but unstable

If you set max depth to -0.02 to slow down a bit?

svenhb, 2019-01-16:
Better solution is perhaps to extend the G38.x command by your own (perhaps .6). I would try:
Adapt grbl:
gcode.c line 173 || (mantissa == 60) and line 809
Add new define to gcode.h line 77
Further changings are on you....
Note: probably you need to implement A/D ISR - no idea if it's possible.

svenhb, 2019-01-16:
Check new update: https://github.com/svenhb/GRBL-Plotter/blob/master/GRBL-Plotter_1233_publish.zip
* Z-adaption also for G0 when applying height map (only to pos. values).
* Add 3rd decimal to Surface scan Max. depth
* Suppress [PRB: message during height scan

wwwlxs, 2019-01-16:
> You need to adapt grbl itself to implement this - perhaps with own G-number as a command. Perhaps M66 (http://linuxcnc.org/docs/html/gcode/m-code.html#mcode:m66).
> Perhaps other grbl implementation already have this - for other controller like arduino duo, stm32?
> Then I could process this value in grbl-Plotter.
Thanks sevenhb， I don't know much about grbl's code, it looks a bit difficult.
I have to continue learning code to understand it.

> If you set max depth to -0.02 to slow down a bit?

it is not much differences

> Better solution is perhaps to extend the G38.x command by your own (perhaps .6). I would try:
> Adapt grbl:
> gcode.c line 173 || (mantissa == 60) and line 809
chang to:
`if（!((mantissa==20)||(mantiss==30)||(mantissa==50)||(mantiss==60))){`
?

809:
`case Motion_Mode_Probe_Toward_no_error:case Motion_Mode_Probe_away_no_error:`

> Add new define to gcode.h line 77
en，line 77   is blank by default

> Further changings are on you....
> Note: probably you need to implement A/D ISR - no idea if it's possible.
Currently I only know how to get the sensor voltage from the Arduino uno A4 iO and calculate it in millimeters then ‘print’ to COM ...

wwwlxs, 2019-01-16:
> Check new update: https://github.com/svenhb/GRBL-Plotter/blob/master/GRBL-Plotter_1233_publish.zip
> 
> * Z-adaption also for G0 when applying height map (only to pos. values).
> * Add 3rd decimal to Surface scan Max. depth
> * Suppress [PRB: message during height scan

Ok, thank you very much.

wwwlxs, 2019-01-16:
It often stops suddenly, but the mouse can move. If you press stop scanning Height, the scan will restart and not continue with the last scan.[image]
[image]
[image]

svenhb, 2019-01-16:
Without real hardware its hard to test for me...
Strange are the different errors: 33, 24 and 2. Perhaps arduino COM is busy and loses characters?
"2","Bad number format","Missing the expected G-code word value or numeric value format is not valid."
"24","Invalid gcode ID:24","More than one g-code command that requires axis words found in block."
"33","Invalid gcode ID:33","Motion command target is invalid."

Do you use original Arduino or 3rd party with CH341 USB?

wwwlxs, 2019-01-16:
Original Arduino UNO for sensor，and 3rd party Arduino Grbl board for plotter 。

[41 more replies]