svenhb/GRBL-Plotter #37: Trouble in some svg file coming from Fritzing

hervema, 2018-11-04:
Hello. 

Very nice project !! 

I'm trying to use it, for my plotter project for which i'm going to use **svg** files generated by Fritzing.

However the tool isn't able to read it properly.

Attached some files : the original svg file, the correct view of this SVG and the view in the tool (last version 1.2.0.1)

Hope you can help me to resolve that in order to use your wonderful project.

Thanks a lot !

Regards,

Hervé
[PrblGRBL-Plotter.zip](https://github.com/svenhb/GRBL-Plotter/files/2545649/PrblGRBL-Plotter.zip)

svenhb, 2018-11-04:
Thanks for the feedback.
I found the reason, it's fixed in the next version (soon).

hervema, 2018-11-04:
Great. I'm looking forward to test that ?? thanks.

Regards

Télécharger Outlook pour Android<https://aka.ms/ghei36>

closed 2018-11-05 by svenhb

hervema, 2018-11-06:
tested and approved ! 👍 Thanks a lot. regards

commit d857d40d (2018-11-05): Fixes #37