svenhb/GRBL-Plotter #38: functions handling the data revceived on com-port

deHarro, 2018-11-07:
A command line should end with \r\n (CR LF) to be recognized as one command.

        private void serialPort_DataReceived(object sender, System.IO.Ports.SerialDataReceivedEventArgs e)
        {   while ((serialPort.IsOpen) && (serialPort.BytesToRead > 0))
            {   rxString = string.Empty;
                try
                {   rxString = serialPort.ReadTo("\r\n");              //read line from grbl, discard CR LF
                    this.Invoke(new EventHandler(handleRxData));        //tigger rx process 
                    while ((serialPort.IsOpen) && (isDataProcessing)) ;  //wait previous data line processed done
                }
                catch (Exception errort)
                {   serialPort.Close();
                    logError("Error reading line from serial port", errort);
                }
            }
        }

        private void handleRxData(object sender, EventArgs e)
        {
            rtbLog.AppendText(string.Format("< {0} \r\n", rxString));
            OnRaiseCommandEvent(new CommandEventArgs(rxString));
            isDataProcessing = false;
        }

_Originally posted by @svenhb in https://github.com/svenhb/GRBL-Plotter/issues/29#issuecomment-418317564_

svenhb, 2018-11-07:
You are right, I will change it next time.

svenhb, 2018-11-10:
It's already implemented in the current version. I check for CR or LF:

```
        private void serialPort_DataReceived(object sender, System.IO.Ports.SerialDataReceivedEventArgs e)
        {   while ((serialPort.IsOpen) && (serialPort.BytesToRead > 0))
            {
                byte[] rxBuff = new byte[serialPort.BytesToRead];
                serialPort.Read(rxBuff, 0, serialPort.BytesToRead);
                byte rxTmpChar = 0;
                try
                {   foreach (byte rxTmpChart in rxBuff)
                    {
                        rxTmpChar = rxTmpChart;
                        if ((rxTmpChar > 0x7F) || (isRealTimeCmd.Contains(rxTmpChar)))  // is real time cmd ?
                        {   rxChar = rxTmpChar;
                            this.Invoke(new EventHandler(handleRxData));
                        }
                        else
                        {   rxTmpString += (char)rxTmpChar;
                            if ((rxTmpChar == '\r') || (rxTmpChar == '\n'))             // end of regular command
                            {   if (lastChar >= ' ')
                                {
                                    rxChar = 0;
                                    rxString = rxTmpString.Trim();
                                    this.Invoke(new EventHandler(handleRxData));
                                }
                                rxTmpString = "";
                            }
                        }
                        lastChar = rxTmpChar;
                    }
                }
                catch (Exception errort)
                {   serialPort.Close();
                    logError("Error reading line from serial port", errort);
                }
            }
        }

```

closed 2018-11-10 by svenhb