svenhb/GRBL-Plotter #39: Performance issue with adaptive path

MarioPerini, 2018-12-11:
Hallo Sven

Danke für diese Software, echt gut gemacht!

Unfortunately i have a big performance issue with grbl-plotter 1.2.1.0 if i load big nc files.
I started using Fusion360 and got a 2.7MB gcode. The software seems not able to handle this.
the grbl-controller or planet cnc can handle the file so there should nothing be wrong.

My Laptop also should handle it with 32GB RAM, i7-8550U, SSD etc.

So is there a known bug on big files?

Best regards
Mario

svenhb, 2018-12-11:
Vielen Dank.
Up to know I never tried such big files, perhaps it's possible for you to upload one here, so I could play around with meaningful data?

MarioPerini, 2018-12-11:
Hi Sven

I send it by mail.

Best regards
Mario

svenhb, 2018-12-11:
Ok, thanks.
I could load the code without problem on Win7 i5-6500 3,2GHz 8GB Ram.
Zooming is a bit slow. No hardware to check a real run.
What is the actual problem "...seems not able to handle this"?

svenhb, 2018-12-11:
At home with my old notebook i5 M460 2,53GHz 4GB RAM I can also load the GCode. But I get a message about unknown code "%" in first and last line (option enabled in Setup - Control - Load GCode).
2D View is much slower than with the other PC :-(

MarioPerini, 2018-12-11:
i have no idea what happens before...  i even did a restart and had the problem again. the plotter hang up und did not respond to clicks. now it works without any issue. I'm really sorry for false reporting.

MarioPerini, 2018-12-11:
[image]

closed 2018-12-11 by MarioPerini