svenhb/GRBL-Plotter #46: Feed errors on slow motions...

kenansatir, 2018-12-30:
There is an speed (feed) error between X and Y axis if your code has slow F commands (for example F10)...
if there is a horizontal line in code, X axis does not use that feed value... moves fast...
But if there is a vertical line in code there is no feed error...

If i type same commands directly from COM CNC window, there is no error...

kenansatir, 2018-12-30:
G54 (Setup - GCode-Header)
M3 S20000 (Start)
G00 X-10.000 Y-10.000  (No problem)
G01 X-10.000 Y10.000 F10 (No problem)
G01 X10.000 Y10.000  (Moves faster than F10)
G01 X10.000 Y-10.000  (No problem)
G01 X-10.000 Y-10.000 (Moves faster than F10)  
M30 (Setup - GCode-Footer)
M30 (Program end)

kenansatir, 2018-12-30:
I have tested your software with versions 1.2.1 and 1.2.2 .... same result

svenhb, 2018-12-31:
How did you solve this issue? Perhaps too low feedrate for grbl?

kenansatir, 2018-12-31:
Too low steps/mm value for the stepper clock... its nonsense but nothing to do...

svenhb, 2019-01-04:
What was the problem?

closed 2019-01-04 by svenhb