svenhb/GRBL-Plotter #54: Errors

Gabse, 2019-01-05:
Hi, I tried to run the new version from Release folder and get the following issue on run:
[image]

After hitting OK the following window appears:
[image]

It can be cleared by hitting ok again and the software seems to work.
When I try to navigate to settings and GamePad I get another error:
[image]

It won’t disappear by clicking OK, and I have to kill the software via Task manager.

Something else: does limiting the working area work at your setup?
In my case it works on X and Y-axis, but on Z it will exceed de work area.

svenhb, 2019-01-05:
During normal setup, the hotkeys.xml would be copied to the correct location.
Now you have to copy it manually from here: https://github.com/svenhb/GRBL-Plotter/blob/master/GRBL-Plotter/hotkeys.xml

svenhb, 2019-01-05:
I copied the missing file to the given location.

closed 2019-01-05 by svenhb