svenhb/GRBL-Plotter #56: Problem with S command

Comet12, 2019-01-06:
Hi Sven,
I encountered a problem with the S command in a nc file an GRBL-Plotter.

Acording the S command in the nc file the spindle should work with 15000 rpm,
but it turned only with the min rpm specified in GRBL (in my case 3000 rpm).
With other G Code Senders, e.g. UGS or bCNC the problem doesn't exist.
I use GRBL V1.1f.

The nc file contains the folliwing gcode:
%
G90 G94
G17
G21
(Planen1)
S15000 M3
G54
G0 X15.3 Y-0.302
Z27.7
Z7.7
G1 Z5.2 F100
G18 G3 X15.1 Z5 I-0.2
G1 X14
X-1 F300
G17 G2 Y1.377 J0.84
G1 X14
G3 Y3.057 J0.84
G1 X-1
G2 Y4.736 J0.84
G1 X14
G3 Y6.416 J0.84
G1 X-1
G2 Y8.095 J0.84
G1 X14
G3 Y9.775 J0.84
G1 X-1
G2 Y11.455 J0.84
G1 X14
G3 Y13.134 J0.84
G1 X-1
G2 Y14.814 J0.84
G1 X14
G3 Y16.493 J0.84
G1 X-1
G2 Y18.173 J0.84
G1 X14
G3 Y19.852 J0.84
G1 X-1
G2 Y21.532 J0.84
G1 X14
G18 G2 X14.2 Z5.2 K0.2
G0 Z27.7
M5

svenhb, 2019-01-06:
Hi Thomas,
I don't think it's a problem of GRBL-Plotter. 
The command is send correctly if you check output of COM CNC window. What grbl on the controller is doing with this command is not in my fault...   
Output in COM CNC window:
```xml
[Start streaming - no echo]
Save last pos.: X=0.840 Y=3.360 Z=5.200
> G90 G94 
> G17 
> G21 
> S15000 M3 
.<Idle|MPos:-3.260,-1.140,5.200|FS:0,1000|Ov:100,100,100|A:S>
> G54 
> G0 X15.3 Y-0.302 
```

Comet12, 2019-01-06:
Hi Sven,
I stripped the nc file until the S  and M command and the result in the
serial connect window is the following:
[image]
I have no idea where the S1000 command is come from...

svenhb, 2019-01-06:
It seems to be a feedback loop...
If spindle is switched off in GUI and you send M3 in COM CNC window, the spindle-checkbox will be set to show spindle is on, unfortunatly this causes the GUI to send the command again - as if you check the checkbox manually...
I have to fix it

Comet12, 2019-01-06:
Ah, now I understand. Yes the 1000rpm comes from the control window. After re-start
it always the same value and not the last used....

Comet12, 2019-01-06:
Awesome! It works, thank you!

BTW: Have you a recommendation how to handle a manual tool change?
At the moment I add a M0 command before M6.

svenhb, 2019-01-06:
If you get GCode from GRBL-Plotter by importing SVG or DXF you can select to add Tx M06 or M0 command.
If you already have GCode I could offer a textbox for text-replacemtent during load of code, like the implemented "Convert Spindle commands" in setup - control - load gcode.
Perhaps two text boxes for search and replacement string...