svenhb/GRBL-Plotter #58: Feature request: Display feedrate and spindle speed in main window

Comet12, 2019-01-30:
Hi Sven,
it would be very useful to see the actual spindle speed and feedrate in the main window.
To display them should be easy, the needed information is also displayed in the serial window.

So long
Thomas

svenhb, 2019-01-30:
Have a look to the new version 1.2.3.9: https://github.com/svenhb/GRBL-Plotter/releases/latest

Override of spindle speed doesn't work with grbl_v1.1f.20170801.hex? Can you confirm?

Comet12, 2019-01-31:
Thank you for your work, I can see now both value!

In the meantime I switch to grbl v1.1g on my uno and mega and spindle speed override
works fine with both. 
BTW: I don't use the pre-compiled hex-file, I compile with Atmel Studio on my own.

svenhb, 2019-01-31:
Thanks for the feedback

closed 2019-01-31 by svenhb