svenhb/GRBL-Plotter #59: Zeroing buttons should use G10 L20 P0 command instead of G92

Comet12, 2019-02-13:
Hi Sven,
the usage of G92 for zeroing of coordinates results in problems with more
than one coordinate-systems (G55, G56, etc.).
The offset invoked with G92 will be applied on all coordinate-systems,
but they are usually used for more than one work-(piece)-coordinate-systems.
With G92 all coordinate-systems are moving and thats not ok.
Instead G92 you should use G10 L20 P0 and everything is fine.

So long
Thomas

svenhb, 2019-02-13:
Funny, I just added a new window to handle coordinate system.
I changed the command as you suggested - I hope I found all for replacement.
Check new test version 1.2.310

svenhb, 2019-02-27:
No bugs reported, seems to work as expected

closed 2019-02-27 by svenhb