svenhb/GRBL-Plotter #60: Closing connection via Close Button should save last position

Comet12, 2019-02-13:
Hi Sven,
closing the connection via Close Button in COM window should save the current position
in the same way as closing the application. 

Regards
Thomas

svenhb, 2019-02-13:
I added this feature to the new test version 1.2.3.10 
https://github.com/svenhb/GRBL-Plotter/blob/master/GRBL-Plotter_12310_publish.zip

Comet12, 2019-02-14:
Saving via Close button doesn't work (old behaviour). Are all relevant files enclosed?

svenhb, 2019-02-14:
I just uploaded the compiled version in a zip file, no single files.

Comet12, 2019-02-14:
OK, but the feature doesn't work.

svenhb, 2019-02-14:
I checked the code right now: I made a mistake with the code change...
I will fix it this evening

svenhb, 2019-02-14:
Please check again https://github.com/svenhb/GRBL-Plotter/blob/master/GRBL-Plotter_12311_publish.zip

svenhb, 2019-02-27:
No bugs reported, seems to work as expected

closed 2019-02-27 by svenhb