svenhb/GRBL-Plotter #61: Camera Index

robZeilinga, 2019-04-16:
**Describe the bug**
When Selecting a "USB Camera" in the second position in the select camera's menu 
I encounter a "specified cast is invalid" exception .
in file : **cameraControlForm.cs**
on the line :   **cameraIndex = (byte)clickedItem.Tag;**  in the 
**private void camSourceSubmenuItem_Click()** event 

to resolve : 

I changed the cameraIndex variable to an int.
(
in the definition of the variable, 
the line throwing the error above, 
and in the properties file 
}

robZeilinga, 2019-04-16:
in further testing, there is a further issue with my resolution. 
the bug still exists, and i'll keep trying to identify the issue.

svenhb, 2019-04-16:
Perhaps your camera is not supported by AForge.Video.DirectShow?
Try this example, if it works, the problem is probably in GRBL-Plotter: http://www.aforgenet.com/framework/samples/computer_vision.html

robZeilinga, 2019-04-17:
found the cause, 
in the line 
           cameraIndex = (byte)clickedItem.Tag;  
I changed this to :  
            cameraIndex = Convert.ToByte(clickedItem.Tag); 
and  in the line 
       selectCameraSource(cameraIndex, cameraResolution);
changed to 
      selectCameraSource(Convert.ToInt16(cameraIndex), cameraResolution);

all working now,  
I have added a commit on the project - will be adding a pull request soon .

closed 2019-04-28 by svenhb