svenhb/GRBL-Plotter #63: drag knife problem

lols, 2019-05-10:
Hi, i set the knife offset and it cut the corners perfectly, but on straight line, the start point does not meet with the end point and leave around 1mm or so. How to over cut slightly so that the start and end point will meet?

svenhb, 2019-05-10:
Thanks for the feedback. Hope my fix works as expected.

closed 2019-05-10 by svenhb

lols, 2019-05-11:
Not fix yet.
The issue here i guess is because the cutter knife is a v shape and different material thickness and knife depth will have different length to start early. If is near the tip it could be 0.1mm or deeper could be 0.5mm.
What i see from your code is for example, the compensate is 0.25, you start earlier by 0.25

Say the start point x,y is 0,0 but because of the knife v shape, it need to start earlier or end later depending on how u want to code so it could be and addition of x,y -1,0 or 0,-1 depend on horizontal or vertical line.
My idea would be to add an additional textbox for user to input the pre cut or post cut additonal length in the setup under drag knife
And another

svenhb, 2019-05-11:
I don't get it.
With pre- and postcut you only mean before a figure starts (not in the middle) and when a figure is ninished?
The additional value will be applied everytime (added to the cutter offset)? Or just replaces the pre- and postcut value?
Thanks for the feedback
Btw there is still a bug when importing DXF and precut at G2/G3 moves...

lols, 2019-05-11:
My bad i figured out the problem, it is the $32=1 laser mode. M3 only activate in G1
Set to 0 and it seems to cut well now for paper. Will do more testing tmr
The pre-post cut, i might have use the wrong word, but I'm trying to say for "post-cut" when it stop at the end letter g last point, it will cut eg another 0.5mm to confirm cut through.
example
for the grbl.svg
the letter g, 
it ends at this line
G01 X13.6403 Y24.0198
it would be good to have a function to cut additional length to confirm it cut through.(over cut it since it is moving in  <- direction)
G01 X13.1403 Y24.0198
my ideal of pre cut/post cut is
eg start point X0 Y0 
Pre cut it with a user defined value, eg 0.5mm in setup
so it will be X-0.5,Y0 or X0,Y-0.5 depend on where it is going, horizontal-> or vertical^
same with post cut
eg original end at X5 Y5 
over cut it by 0.5mm
it will be X5.5 Y5 (-> direction )or X5 Y5.5 (^) or down X5 Y4.5

commit bf9d1d43 (2019-05-10): fix #63