svenhb/GRBL-Plotter #66: Tool Length Offset Support

Comet12, 2019-08-09:
Hello Sven,

does GRBL Plotter support automatic Tool Length Offset via grbl hardware trigger?

It would be a great future, because my cnc job's have a lot of tool changes.

Thanks in advance, Regards

Thomas

svenhb, 2019-08-09:
Yes it's possible - shown in this video https://youtu.be/x5UTHpgsfII e.g. on 0:30 and 1:40.
You need to define your scripts.
Check the files script_pen_pickup.nc, script_pen_remove.nc, script_pen_select.nc and script_probing_gripper.nc in subfolder _misc, which where used in this video.
Just select the coresponding scripts in setup - tool exchange.
The 'magic' lines are:
```
G38.3 Z-10		(probe toward tool length sensor, stop on contact)
G43.1 Z#PRBZ	 	(Offset Tool)
```

#PRBZ is a variable, holding the probing result (which is usually not the stopping position, because of decelaration of the axis).

I should make a description inside the wiki....

Comet12, 2019-08-10:
Hi Sven,
thanks for the reply.
I do always the tool change manually, how could the solution be?
1. With the first tool in spindle I probe the zero workpiece point (X0, Y0, Z0) without tool length sensor manually with a feeler gauge.
2. After changing the tool I would do a probe with the tool length sensor, but how and
where do I define the height of the tool length sensor (e.g. 32mm)?
3. From my point of view I need an initialisation with the first tool, correct?
Regards
Thomas

svenhb, 2019-08-10:
You need real homing to be able to use machine coordinates. Then use G53 to move to length Sensor.

svenhb, 2019-08-11:
Probably it's also possible without real homing:
1) probe your actual tool on a fix or movable sensor (with fix distance to machine floor)  
2) set zero of your workpiece  (X0, Y0, Z0)  
3) do your work with first tool
4) change tool, do probing of tool length the same way as in 1) (same distance to machine floor)

grbl then takes account of tool lengths difference

Comet12, 2019-09-01:
Hi Sven,
sorry for the late response.
I now use a tool length sensor and it's amazing how it helps ;-)
I add two functions on two command buttons and to configure it
was much easier than expected. One button to move to the tool change
position and one to do the tool length measuring.
Thank you for your support!
Regards
Thomas

svenhb, 2019-09-03:
Thanks for the feedback, 
did you buy a sensor or did you made your own?
Regards
Sven

svenhb, 2019-10-03:
Fixed by use of custom buttons

closed 2019-10-03 by svenhb