svenhb/GRBL-Plotter #67: Index was outside the bounds

raminrajabi, 2019-08-12:
[image]

when I want to connect to my Arduino port and after some command comminution I got this error message in log text box: out of bounds

in line 817:
 grbl.getPosition(dataField[4] + "," + dataField[5] + "," + dataField[6]+" ", ref posWork);

my dataFiled only has 4 items but in the line above it's want to read the item 5 and 6;

the command is : Idle,WPos:0.000,0.000,0.000

svenhb, 2019-08-12:
I will check. Perhaps you can upgrade to grbl version 1.1?
https://github.com/gnea/grbl/releases/tag/v1.1f.20170801

raminrajabi, 2019-08-12:
I downloaded the latest version of grbl and compiled it and I've got this message but when I downloaded the precompiled hex file and using it the problem is gone.
now it's working perfectly
thanks.

svenhb, 2019-08-13:
>I downloaded the latest version of grbl and compiled it 
Can you give me the link to the source you used?

raminrajabi, 2019-08-13:
[grbl.zip](https://github.com/svenhb/GRBL-Plotter/files/3496286/grbl.zip)

a used source with compiling of Arduino UNO

svenhb, 2019-10-03:
Found weak code, add a check if index is within bounds

closed 2019-10-03 by svenhb