svenhb/GRBL-Plotter #68: Dpi Problem nach dem Laden dll

maxx04, 2019-09-10:
**Describe the bug**
Auf dem Hochauflösenden Bildschirm 200% Vergrößerung eingeschaltet.
Beim Start alles in Ordnung. Wenn lade Beispiel Bunny.dxf oder anderen .dxf, dann nach dem Aufruf **processEntities(dxfEntity);** wird dll DXFLib geladen, dann Skalierung geht verloren.
Und MainForm sieht klein aus.
[image]

Hab versucht die Korrektur in Program.cs -> alles läuft aufanander. Suche weiter auf Lösung.
 ```
static void Main()
        {
            //if (Environment.OSVersion.Version.Major >= 6)
            //    SetProcessDPIAware();

            Application.EnableVisualStyles();
            Application.SetCompatibleTextRenderingDefault(false);
            bool newInstance;
            using (var mutex = new Mutex(true, "Deine_Anwendung", out newInstance))
            {
                if (newInstance)
                    Application.Run(new MainForm());
            }
        }
        //[System.Runtime.InteropServices.DllImport("user32.dll")]
        //private static extern bool SetProcessDPIAware();
        //Application.Run(new MainForm());
```
``

svenhb, 2019-09-10:
Zu diesem Problem habe ich z.Zt. leider keine Idee zur Lösung

maxx04, 2019-09-13:
Eigentlich das ist keine Lösung. Ich habe Program.cs geändert wie oben, Positionsberechnungen in MainForm.cs auskommentiert. Hab versucht die Tabellen umzuordnen. Ergebniss kannst du auf dem Git anschauen <<https://github.com/maxx04/GRBL-Plotter>>

svenhb, 2019-10-03:
Zur Zeit keine Lösung in Sicht, da ich das Problem nicht nachstellen kann

closed 2019-10-03 by svenhb