svenhb/GRBL-Plotter #69: CNC2418 AND SIMILAR WITH WOODPECKER BOARD

ivahru, 2019-09-24:
Similar case was already discussed but not solved yet. I have CNC2418 with Wodpecker2.6 board with updated GRBL1.1. Everything works perfectly with CANDLE latest edition. Problem is that Wodpecekr boards have probably directly connected RESET to DTR signal from CH340. Arduino uses RC derivation circuit and then static value on DTR pin does not play a role. In Wodpecker yes. Symptoms are that after start of GRBL Ploter spindle starts and machine does not react to any command. GRBL Ploter SW hangs and serial line window shows message as there is no received response. It cannot be as controller is held in RESET mode. I tried to modify the code and added control of DTR signal. Then machine and SW work perfectly. Pls add this modification to next code issue:

 private bool openPort()
        {
            try
            {
                serialPort.PortName = cbPort.Text;
                serialPort.BaudRate = Convert.ToInt32(cbBaud.Text);
                serialPort.Open();
// This is DTR init - then machine works again
                serialPort.DtrEnable = true;
                serialPort.DtrEnable = false;
                rtbLog.Clear();

svenhb, 2019-09-24:
Thank you to show also the solution.
A new version is in progress (with your solution) and will be released soon.

ivahru, 2019-09-25:
Hello ,
in attached excell is math how to get transformation matrix from three 
points (i.e. camera scans three pads on PCB ) .

Regards,
Ivan

svenhb, 2019-10-03:
Implemented your suggestion in Update 1.3.0.0

closed 2019-10-03 by svenhb