svenhb/GRBL-Plotter #70: Camera & pattern restitution

ivahru, 2019-09-24:
I must say I was happy to see this SW first as this is something I'd like to have. I did some three years ago similar tool to drill etched PCBs using modified 3D printer as PCB drill machine. Added small camera USB microscope to drill head and added some code to existing control SW ( Reprap ) in C language. Finaly I developed method how to perform restitution as simple as possible to set. Unfortunately mechanics of printer was not enough stable for my usage ( rubber belts). Method used in GRBL Plotter seems to be quite compllicated. Then I'll describe my own as suggestion for improvement:
1) put PCB to the working table ( generally any position but roughly along axes is more precise)
2) Drill manually tho hole using manual control of SW somewhere on the edge of PCB.
3) Machine knows the coordinates. Move camera crosshair over this hole. Push button - difference of camera and tool coordinates is offset vector.
4) Then you need to move camera over three pads on PCB, ideally in right ( or close to right) . Each pad to the crosshair.  In this code is possible graphically choose every point ( my tool had number  input where I wrote real coordinates of marked pad) . 
5) From those data you can compute transformation matrixes for offset & zoom ( separately X and Y ) . This is important as the PCB pattern has different scales in X and Y directions , but matrix solves everything.
6) Then drill can start

My implementation was on the side of GRBL firmware inside machine by extension of several special G-codes, and later adding my own control SW in VBA. No integrated camera, just using existing microscope SW to dislplay PCB.

Your method is like similar but I can see some complicated camera calibration ( rotation feature is GOOD as camera is never precisely assembled) and trimming. Control is not very intuitive. 
Important is also to grab pads by camera when tool is in Z=0. You have that compensation of Z but in home conditions is ever difficult to have real parallel axes of tool and camera and anyway the focus is not good if too high above PCB. 

During my several tries the SW also crashed ( with question if I want to leave it ). Reason - just wrong click on pad in camera view. Looks like some division by zero or something out of bounds.
Regards,
Ivan

svenhb, 2019-09-24:
Thanks for the feedback. I have not tried camera functionality for many months (years?).
I will check/try your suggestions.

ivahru, 2019-09-24:
I'll find my old code so you can try to implement it.
Regards,
Ivan

svenhb, 2019-10-03:
Sorry found no attachment

closed 2019-11-24 by svenhb