svenhb/GRBL-Plotter #72: is this normal in gcode? please help

oraherb, 2019-10-07:
why my gcode that i created is  like this  did i do anything wrong?

oraherb, 2019-10-07:
is this gcode is correct  i cant see the image all i can see is green line?? [image]

svenhb, 2019-10-07:
You can click on the folded code '+', e.g. line 12 to see the real gcode.
Send me your picture or graphics, then I can check.

oraherb, 2019-10-08:
@svenhb  i have a problem when i drag and drop an image to gbrl plotter then  create gcode after that  the gbrl plotter is loading i mean  it doesnt respond   did i do anything wrong with the setting? because  when i first use this gbrl plotter is it not like this after i create an image gcode  it doesnt load    and hang like this now how can i reset it default settings

oraherb, 2019-10-08:
@svenhb hey bro nothing happen still like this and my hardware is not creating that image it just follow all that line no fry image was created what is wrong ?[image]

svenhb, 2019-10-08:
Please make screenshots of Setup-graphics-import tabs

oraherb, 2019-10-08:
@svenhb bro heres a video i uploaded it on youtube what my cnc doing (https://youtu.be/7B1wket5BYM)

oraherb, 2019-10-08:
[image]
you

oraherb, 2019-10-08:
[image]

svenhb, 2019-10-08:
I have no time to watch video...
In image-window, select use-case plotter, then restore your servo-pwm settings in setup-graphics import.
What size and resolution you selected on imgage-correction?

oraherb, 2019-10-08:
here is path import [image]

oraherb, 2019-10-08:
heres g code generation 
[image]

oraherb, 2019-10-08:
here g code modification 
[image]

oraherb, 2019-10-08:
use cases
[image]

svenhb, 2019-10-08:
You should switch off all checkboxes, but at least the marked:
[image]

svenhb, 2019-10-08:
I would use this settings:
[image]

oraherb, 2019-10-08:
@svenhb bro do u know a website were i can read  and understand all the setting and values of gbrl   plotter program?

svenhb, 2019-10-08:
I thought most of the checkboxes are self-explaining...
Perhaps I can make a manual based on your questions...
Which setting you are not understanding?

oraherb, 2019-10-08:
@svenhb bro its i use modified grbl its working and i replace the drv8825 with L293D IC well im not satisfied because i didnt learn how  to make the gbrl plotter work well i have to learn it to be satisfied
[image]

svenhb, 2019-10-08:
Is this picture made with grbl-plotter? So finally you get it to work - very good!
Yes hardware can be tricky, perhaps too weak powersupply for the drivers? Perhaps wrong calculation for "step/mm" and wrong calculation with microsteps, set at the driver...
It should also work with the drv8825.

oraherb, 2019-10-09:
@svenhb  hey bro yes it my pen plotter who  made that drawing in above picture   but i have to use this  modified gbrl to make it work  and processing t3 to send gcode

#include <Servo.h>
#include <Stepper.h>

#define LINE_BUFFER_LENGTH 512

const int penZUp = 80;
const int penZDown = 40;

const int penServoPin = 6;

const int stepsPerRevolution = 20; 

Servo penServo;  

Stepper myStepperY(stepsPerRevolution, 2,3,4,5);            
Stepper myStepperX(stepsPerRevolution, 8,9,10,11);  

struct point { 
  float x; 
  float y; 
  float z; 
};

struct point actuatorPos;

float StepInc = 1;
int StepDelay = 0;
int LineDelay = 50;
int penDelay = 50;

float StepsPerMillimeterX = 6.0;
float StepsPerMillimeterY = 6.0;

float Xmin = 0;
float Xmax = 40;
float Ymin = 0;
float Ymax = 40;
float Zmin = 0;
float Zmax = 1;

float Xpos = Xmin;
float Ypos = Ymin;
float Zpos = Zmax; 

boolean verbose = false;

void setup() {
  Serial.begin( 9600 );
  
  penServo.attach(penServoPin);
  penServo.write(penZUp);
  delay(200);

  myStepperX.setSpeed(250);
  myStepperY.setSpeed(250);  

  Serial.println("Mini CNC Plotter alive and kicking!");
  Serial.print("X range is from "); 
  Serial.print(Xmin); 
  Serial.print(" to "); 
  Serial.print(Xmax); 
  Serial.println(" mm."); 
  Serial.print("Y range is from "); 
  Serial.print(Ymin); 
  Serial.print(" to "); 
  Serial.print(Ymax); 
  Serial.println(" mm."); 
}
[... 255 lines cut ...]
  }

  
  delay(LineDelay);
  
  Xpos = x1;
  Ypos = y1;
}

void penUp() { 
  penServo.write(penZUp); 
  delay(LineDelay); 
  Zpos=Zmax; 
  if (verbose) { 
    Serial.println("Pen up!"); 
  } 
}
void penDown() { 
  penServo.write(penZDown); 
  delay(LineDelay); 
  Zpos=Zmin; 
  if (verbose) { 
    Serial.println("Pen down."); 
  } 
}

oraherb, 2019-10-09:
@svenhb  bro can you help me with this  how can i set this gcode to 0 position  because my plotter needs to run to fat to start the drawing
[image]

svenhb, 2019-10-09:
Just click here and apply offset
[image]

oraherb, 2019-10-10:
@svenhb bro is there any  way that i can generate gcode other than inkscape and   gbrl plotter  or easel i mean gcode for pen plotter     that have  pen up and down

oraherb, 2019-10-10:
because gbrl plotter doesnt work  im using processing t3 to send gcode to arduino  i try  making gcode in  gbrl plotter then export the gcode and open it in processing T3  but it doesnt work and i try inkscape it doesnt work also thats why im trying to find something that will work on the modified gbrl that i use bro

svenhb, 2019-10-10:
Very complicated... Why not using GRBL-Plotter to send the code?
Use a 'normal' grbl firmware like https://github.com/svenhb/GRBL-Plotter/blob/master/grbl_v1.1f_Servo.hex
I tested it with my egg-bot - it works...

svenhb, 2019-10-11:
The grbl version I mentioned, uses step and dir signal for the A4988 (or similar) drivers.
Why using L293 (is'nt it very outdated?), when you can get µC and drivers for 10€ - just check ebay for "Uno-R3 CNC-V3-0-shield-4x-A4988"

svenhb, 2019-11-24:
Question is answered, I think.

closed 2019-11-24 by svenhb