svenhb/GRBL-Plotter #73: Spindle CW is unreliable

cassidywilliams, 2019-10-09:
Was using the same configuration in 1.25 and unable to see the same results in 1.3.0.0. Expected behavior: ability to check Spindle CW box to move servo (pen up/down on a x,y plotter). The box does not remain checked and the servo goes up and down in a seemingly random manner.

Settings:
pen up: 0
pen down: 30

spindle speed 25

svenhb, 2019-10-09:
Are you using the laser mode ($32=1)? Then spindle only stays on, when G1 movement was performed.
Read here: https://github.com/gnea/grbl/wiki/Grbl-v1.1-Laser-Mode

svenhb, 2019-10-09:
The check box 'Spindle CW' in main GUI sends M3 or M4 for 'on' and M5 for 'off'.
To use servo you need to send M3 S0 and M3 S30 (as you wrote above).
Use a custom button to set and send this commands.

cassidywilliams, 2019-10-09:
This works. Thanks for the fast help.

closed 2019-10-09 by cassidywilliams