svenhb/GRBL-Plotter #74: "Save GCode" causes error

theC4Labs, 2019-10-10:
"File -> Save GCode" Results in a dialog box with the following message:

Filter string you provided is not valid. The filter string must contain a description of the filter, followed by the vertical var (|) and the filter pattern. The string for different filtering options must also be separated by the vertical bar. Example: "Text files(*.txt)|*txt|All files (*.*)|*.*"

Tested on v1.3 and v1.25

Wish I knew how to fix this myself, but I am still an beginning programer. Thanks for the great work!

svenhb, 2019-10-11:
Ups, not tried for longer time.
I think the reason is there are two separators                  V
            sfd.Filter = "GCode (*.nc)|*.nc|GCode (*.cnc)|*.cnc||GCode (*.ngc)|*.ngc|GCode (*.gcode)|*.gcode|All files (*.*)|*.*";   // "GCode|*.nc";

theC4Labs, 2019-10-15:
I assume it will be a easy fix, but understandably something you have not looked at for a while :)

Good job again, I can tell you put a lot of work into making this for the community. I certainly appreciate it.

commit d0840403 (2019-10-23): Update 1.3.0.1 fix #74
closed 2019-10-23 by svenhb