svenhb/GRBL-Plotter #75: Vinyl Plotter With the Z axis is an electromagnet?

GioLangLe, 2019-10-26:
Thanks for your great project. I want the Z axis to work with an electromagnet. What do I have to fix? Or must the Z axis be a Servo motor?

svenhb, 2019-10-26:
Just connect your magnet as a spindle. In Setup - Graphics Import - G-Code generation select Spindle on/off - perhaps you need to set Spindle speed to a higher value - if PWM is enabled in your controller. The Spindle (your magnet) will be switched on on 'Pen down' and switched off on 'Pen up'. Note also the 'Delay after spindle on' to give the magnet time to reach and position.
[image]

svenhb, 2019-11-24:
Question is answered, I think.

closed 2019-11-24 by svenhb