svenhb/GRBL-Plotter #76: Russian lang

Phoenix-I, 2019-10-28:
Hello. Yesterday I wanted to offer an idea, today I got a release.

I would like to help make the Russian translation better.

I have not worked with such projects before, but as a native speaker, I could just rewrite the correct translation.

Email.
phnxmmrtl@gmail.com

FelisucoVFR, 2019-10-28:
Hi,

I can try to polish the Spanish too.

Regards

svenhb, 2019-10-28:
Would be nice to get corrections about translation.
I used this tool to edit resx files: https://github.com/HakanL/resxtranslator
Unfortunatly for one resx (MainForm.resx) a workarround is needed (check https://github.com/HakanL/resxtranslator/issues/38) because it can't be saved.

Or you attach screenshot with needed change...

Phoenix-I, 2019-10-28:
> HakanL/resxtranslator#38

Some thing like this?

///////////////////   1
1 - Открыть
2 - Сканировать	
3 - Послать / Отправить
4 - Текущее состояние
5 - Настройки GRBL
///////////////////   2
6 - Открыть
7 - Перезагрузка
////////////////////  3
8 - Тревога 
9 - Закрыть
///////////////////    4
10 - Прогресс
11 - Снять тревогу
12 - Тревога  (if green "Готов" or "Ожидает", if yellow "Работает")
13 - Нажмите "Снять тревогу" для работы
14 - Смещение по Х
   - Смещение по Y

15 - Обнулить X , Обнулить Y  ... etc 
16 - Спарвка
17 - Контроль/Перемещение
18 - Шпиндель
19 - Скорость
20 - Охладитель
21 - Инструмент в шпинделе
22 - Переместиться в XY=0
23 - Остановить перемещение
24 - Переместиться в локальный 0
25 - Дом (аппаратный 0)
26 - Перезагрузка контроллера
27 - Оставока при отажатии кнопки
28 - Установить новую координату

//////////////////////// 5
29 - Скорость ШИМ
30 - Рабочая подача (better) / Рабочее перемещение (maby)
31 - G-code Заголовок
32 - G-code Конец
33 - Время разгона Шпинделя
34 - Z подача 
35 - Z безопасная высота
36 - Z гравировка

svenhb, 2019-10-28:
I assume you wrote the numbers into screenshots? Please upload the screenshots also...

Phoenix-I, 2019-10-28:
Strange, I attached. Everything in zip

svenhb, 2019-10-28:
Perhaps you need to upload directly in GitHub webpage?

FelisucoVFR, 2019-10-28:
Hi,

Can be possible getting a kind of Excel file with the English and actual Spanish/Russian?

At least for me would be lot easier.

regards

svenhb, 2019-10-28:
Yes, would be ok, but please form related and control related - hope there are not so much words?
E.g. serial-COM form, 'open' button, text in es, ru...

Phoenix-I, 2019-10-28:
Sorry, I thought that the letter to me, and not through the github service. Apparently this file is not attached.

[rus.zip](https://github.com/svenhb/GRBL-Plotter/files/3779409/rus.zip)
[image]
[image]
[image]
[image]
[image]
[text.txt](https://github.com/svenhb/GRBL-Plotter/files/3779410/text.txt)

svenhb, 2019-10-28:
@Phoenix-I Much better... :-)

Phoenix-I, 2019-10-28:
If this fine, later make more :)))

svenhb, 2019-10-28:
Some general questions
- any way to shorten long words? E.g. Pic. 1 "Reset"
- Should GRBL-status be translated? Pic. 3 "Alarm"

Most of the words are too long for Pic. 4

Phoenix-I, 2019-10-28:
I think about it.
Some English words, sound bad in Russian. For example, unpressed (the norm) is "Otjata" (Отжата) - it cuts the ear, We do not say so, if only "in the countryside." But "STOP run on mouse up" is horrible :))) in any language, and long.

At the same time, the meaning is very important.

This is a great starter product. I would like the settings to be clear to the newcomer.

Alarm - here is a common noun and international.
You can leave it as it is :). It is possible for symmetry with "Ready" and "Work" (Run, running).

I think my English is terrible too :)))

FelisucoVFR, 2019-10-28:
I was spending a few minutes on that and I agree with you. In Spanish we also prefer keep some English words. I would maintain some.
Working a little on that later today
👍🏻

svenhb, 2019-10-28:
@Phoenix-I  it's possible to write 'Смещ. X' and have 'Смещение X' inside the tool tip?
Some controls I could increase in size and move a bit, most not.

svenhb, 2019-10-28:
I uploaded some updates...

Phoenix-I, 2019-10-28:
Yes, it good.
In next time, I'll give normal and short words. You can use which need in
place.

FelisucoVFR, 2019-10-28:
Svenhb, do you want me to add the Spanish part here too or opening a new thread?
I don't want to make any dissorder.

svenhb, 2019-10-29:
@FelisucoVFR please open a new thread - it's more clear.

svenhb, 2019-11-24:
No more feedback about translation...

closed 2019-11-24 by svenhb

Phoenix-I, 2019-11-24:
Sorry, a lot of work :( even at the weekend. 

I found some more translation errors, I will try to make a report as soon as possible.

There is also a problem with circles. the circle figure, when generating the G code, does not receive a command, neither to open nor to close "S".

And the problem is if the line is made spline - the file does not open says an error.

A little later I will give details and examples.