svenhb/GRBL-Plotter #78: Spanish language corrections

FelisucoVFR, 2019-10-29:
Hi,

I have a few suggestion to correct the Spanish tags.

[image]

[image]

[image]

[image]

[image]

[image]

1 - "Conectar" (Open) or "Desconectar" (Close)
2 - "Archivo"
3 - "Modificación G-Code"
4 - "Control / Mover"
5 - "Parar al soltar el ratón"
6 - "Transmisión de G-Code"
7 - "Coordenadas (Trabajo/Máquina)"
8 - "Dimensiones G-Code"
9 - "Herramienta en husillo" (Option: "Herr. en husillo" or "Herr. en Spindle")
10 - "Vel." or "Velocidad" or just "Speed" as it is, everybody understand this.
11 - "Refrigerante"
12 - "Pausa"
13 - "Continuar"
14 - "Mostrar informe de estado"

15 - "Abrir archivo"
16 - "Abrir archivo reciente"
29 - "Ajustes"

17 - "Girar -90º (en sentido horario)"
18 - "Rotar ...º"
19 - "Diámetro eje rotativo"

20 - "Sistema de coordenadas"
21 - "Control DIY" (Not sure how to translate this nor what this do...)
22 - "Control 2º GRBL"

23 - "Área de la máquina"
24 - "Tabla de herramientas"

25 - "Acerca de"
26 - "Versión"
27 - "Ruta del programa"
28 - "Comprobar actualización" or "Actualizar"

Just as a start point. I need more free time to enter into the menus, specially the Settings one.

Thanks

FelisucoVFR, 2019-10-29:
Sorry about the "bug" label. No idea how to change it to "enhancement"

svenhb, 2019-10-29:
I made the changes...
https://github.com/svenhb/GRBL-Plotter/blob/master/GRBL-Plotter_1311_es_publish.zip

FelisucoVFR, 2019-10-29:
@svenhb Great, I will try to continue with the rest soon.
Thanks

svenhb, 2019-11-24:
No more feedback about translation...

closed 2019-11-24 by svenhb