svenhb/GRBL-Plotter #80: Dxf file problem

lianzaozi, 2019-11-19:
I tried the parsing function of the dxf file. The spline and ellipse in the current dxf file cannot be parsed. In addition, the moving path display function can be made to move the path display area by pressing the middle mouse button to move the mouse.

svenhb, 2019-11-19:
You may have an exa,ple for me? I just work with LibreCAD, and had no problems...

lianzaozi, 2019-11-19:
[Spline ellipse.zip](https://github.com/svenhb/GRBL-Plotter/files/3863104/Spline.ellipse.zip)

I just tested it with this dxf file.

svenhb, 2019-11-19:
I can't get your ellipse information out of the library I use (https://github.com/mkernel/DXFLib)
So I can't fix it - sorry.
You should try to use arc instead...

svenhb, 2019-11-19:
With a repaired DXFLib.dll I can get the Ellipse information, now I need to implement the ellipse

lianzaozi, 2019-11-20:
You can try using netdxf.dll to read dxf files(https://github.com/haplokuon/netDxf)

svenhb, 2019-11-20:
Now I get the ellipse information, but I need to implement the transform to g-code

svenhb, 2019-11-24:
Should be fixed in Version 1.3.2.3

closed 2019-11-24 by svenhb