svenhb/GRBL-Plotter #81: Display area zoom problem

lianzaozi, 2019-11-20:
[image]
This is what it looks like when you open the file.
[image]
After scrolling the mouse wheel to zoom in, the display area is in the upper left corner, resulting in no visible at all, and there is no way to move the display area.

lianzaozi, 2019-11-20:
At present, the parsing of the spline is also problematic, and the appearance is different from the actual one.

[image]
[image]

Note that this is not the same as the above picture.
In addition ,the accuracy of parsing the spline curve
is not high enough.
You are parsing the spline curve into small line segments.

The accuracy is not high enough.

lianzaozi, 2019-11-20:
This is the dxf file for the test spline
[Spline1.zip](https://github.com/svenhb/GRBL-Plotter/files/3866955/Spline1.zip)

svenhb, 2019-11-20:
Zooming: mouse position is center of zooming - sometimes strange - I can't fix.
Spline: In know there is a weak point I will fix soon.

lianzaozi, 2019-11-20:
OK。
You can refer to the algorithm or idea of dxf2gcode software.(https://sourceforge.net/projects/dxf2gcode/)

svenhb, 2019-11-24:
Spline Import should be fixed in Version 1.3.2.3

closed 2019-11-24 by svenhb