svenhb/GRBL-Plotter #82: How to use the seccond GRBL ?

LeoVerheyden, 2019-11-22:
Is there a Tuto available ?  How to take control off the 2 GRBL tool change ?
I tried with TwM06 but I still sty in the first.

svenhb, 2019-11-22:
Commands for 2nd GRBL will be introduced via special formatted remarks in GCode e.g.: (^2 G90 X2) to move 2nd GRBL to position X2.

Video: https://youtu.be/fvYWyE2GBsg

And check the scripts in data/scripts   script_pen_select, ..._remove, ..._pickup.

Hope that helps...

LeoVerheyden, 2019-11-30:
Thanks it works well. How do I put on zero  G54 in seconds arduino ( com tool changer) ?

svenhb, 2019-11-30:
Should work in the same way: "(^2 G54)" or manually via COM-window of 2nd grbl.
To set set G54 to zero you may use "(^2 G10 L2 P0 X0 Y0)" P0 = active coordinate system, or use P1 to set G54 explicite..
Better read this: http://linuxcnc.org/docs/html/gcode/g-code.html#gcode:g10-l2

LeoVerheyden, 2019-11-30:
Thank you

Leo Verheyden

svenhb, 2020-01-07:
Question was answered - hopefully with success ;-)

closed 2020-01-07 by svenhb