svenhb/GRBL-Plotter #83: Bug, and what to improve

Wdespontin, 2019-11-26:
First I want to congratulate the project, great congratulations, I accompany you from the first versions of GRBL Ploter

bug:
Keyboard and Control
acceleration by the control not to let go of the dorional, has to touch again to stop, this also happens if you hold down the keyboard arrow for a long time

Does not recognize joystick acting on windows (USB)

When starting the code, starts the axis mechanism and no longer turns off when finishing the code, the previous version was perfect in this part

there is an error on the surface when you apply the generated map is floating 3 mm above the cutting base

What to improve (suggestion)

joystick to be used what is already working on Windows (devices)

Missing translation into Portuguese (Brazil) would be excellent!

Being able to position the image in preview

Removing the coordinate frame that appears above the image to be machined makes it difficult to see the progress of the machining process.

If you stop the software by pressing (stop), automatically stop the spindle motor

svenhb, 2019-11-26:
About the virtual joystick inside the GUI:
There is an internal timer, sending the corresponding move-command (setup-joystick) every 0,5 second if mouse button is pressed.
If speed is too low,  movement is not finished after 0,5 sec. the next command appears and will be set into the grbl-command-stack. It seems grbl is not responding on mouse-up.
Therfore I add the checkbox below the virtual joystick, to send the Stop-Jogging command.
https://github.com/gnea/grbl/wiki/Grbl-v1.1-Jogging
I don't see a bug, it is just setup of the virtual joystick control in setup.

About real joystick: I don't have one, will try to implemtent without be able to check.

svenhb, 2019-12-07:
As described above: for me it sounds like my wanted behavior...

closed 2019-12-07 by svenhb

svenhb, 2019-12-08:
Here I add portuguese language:
https://github.com/svenhb/GRBL-Plotter/blob/master/GRBL-Plotter_1325_publish.zip

Wdespontin, 2019-12-08:
Good afternoon, once again congratulations, this software is very good,
congratulations, I already liked the program in English, now in Portuguese
much more, I still do not understand how to use a joystick controller in
the program, but all right, this program trochoidal mill too thanks and
good sunday

*Att; *
*Valter*

svenhb, 2019-12-08:
Thanks. 
What kind of joystick you try to use? 
Did you notice any reaction in [Setup - GamePad] when using your joystick?

svenhb, 2019-12-12:
I bought a 2nd hand Microsoft joystick (Side Winder) and it works...

Wdespontin, 2019-12-13:
good morning you have Watssap would be easier to communicate and exchange
information, follow my +55 (15) 997184931

As for the control I use a PS3 but I do not know how to configure in the
GUI if you can help me thank you, I am using 2 programs so that it can
appear the mouse cursor on the screen and I can in GRBL Ploter and bring
the same to the virtual control and able to move the axes

yes I can help you in the translation I will pass you what to correct in
portuguese brazil

I await your contact soon

*Att; *
*Valter*

svenhb, 2019-12-13:
Did you follow the instructions, how to use a PS3 controller on windows-PC?
For me it sounds like: it's easier to buy an original PC-Joystick (I got mine in a set of thee for 10€):  
Goolge said: 
- press the center button of the controller to switch on, 
- connect controller with PC
- Download driver e.g. from here  ScpToolkit Webseite  and install.

svenhb, 2019-12-13:
Note: a real Joystick needs to be setup in [Setup - GamePad]. (I should rename it)
There is a checkbox on the upper left to enable usage.

Wdespontin, 2019-12-15:
I was able to install the PS3 joystick works 100% on Windows 7 32-bit,
works fine also on GRBL Ploter, analog directional XY and buga when not
responding sometimes, just giving Hard Reset that works again but buga is
very easy

*Att; *
*Valter*