svenhb/GRBL-Plotter #84: Some questions and suggestions

lianzaozi, 2019-11-26:
Hello, what precision did you convert the spline curve and ellipse into small line segments? There is no setting in the software that can set the conversion precision. Will it increase later?
Can I add a function to save a .nc file as a dxf file?

svenhb, 2019-11-26:
I use float and double precision internally.
DXF-spline accuracy is using the value from  setup - Graphics import - G-Code modification 
[image]
I should seperate it and movit to "Path import" tab.

>Can I add a function to save a .nc file as a dxf file?

Yes, why not...

svenhb, 2019-11-26:
DXF-Spline is using this value:
[image]

lianzaozi, 2019-11-27:
Thanks, I am looking forward to adding the ability to save .nc files as dxf files

lianzaozi, 2019-11-27:
I hope you can write a detailed software instruction document and put it in the program folder. Because sometimes how to set some parameters, what does it mean, and how to use some functions is not well understood. With detailed documentation, I think it will help us better understand and use

svenhb, 2019-11-27:
You wrote "Can I add...", so I supposed to get code from you to implement....
I don't such a function, why do you need it? Just do your drawings as DXF (I use LibreCAD) and ready!?

I thought tool tips would be enough... What detailed docu. you need? Use case related?

lianzaozi, 2019-11-27:
Thank you very much for your answer.

When I tested the dxf spline curve subdivision accuracy, I changed the number in the box below to 4 as if nothing had changed, and the number would automatically change to 12, and clicking the Apply button would not work. I changed the number in the box related to the ellipse subdivision to 0.1 and the software did not change. Are these parameters ineffective?

The GRBL plotter software spline curve and ellipse subdivision small line segment G01, compared with dxf2gcode, the software's approach is to convert the spline curve to G02 G03 G01, and the ellipse to G02 G03. I feel that the method of dxf2gcode software is converted into a circle The arc can fit the original shape of the curve to the greatest extent. This is just my personal opinion, maybe the GRBL plotter software has other considerations. In addition, the number of gcode lines generated by the same dxf file dxf2gcode software is more than twice that generated by GRBL plotter software. According to the truth, the more the number of gcode lines, the finer the curve.

lianzaozi, 2019-11-27:
[image]
Modify it to the number 4, click the Apply button, and then re-open the interface, this number will automatically become 12
[image]
After changing to 0.1, the number of Gcode lines and the ellipse trajectory seem to be unchanged.

svenhb, 2019-11-27:
After changing any setting you need to re-import ("Reload SVG, DXF") the file.
I could not find an algorithm to convert spline into arcs, so I convert to line-segements.

>the more the number of gcode lines, the finer the curve.

Only if both codes only uses G1, with G2 / G3 you would get finer resolution with less code-lines.

svenhb, 2019-11-28:
With 4:
[image]

With 12:
[image]

svenhb, 2019-12-07:
Setting is available as shown

closed 2019-12-07 by svenhb

lianzaozi, 2019-12-12:
GRBL-Plotter software splines and ellipses are transformed into small line segments. Setting the subdivision function still has no effect. After setting the values, click the Apply button, and then click the "Reload SVG, DXF, Drill file" button to read the same dxf file. The number of gcode lines has not changed, and the conversion precision number I set has been restored to the software default number. I feel that this number does not seem to have any effect on the conversion precision.

svenhb, 2019-12-12:
I use your 'Spline ellipse.dxf' example:
for me it works...
[image]

But I have to figure out why this doesn't work as expected:
[image]