svenhb/GRBL-Plotter #85: Reading file problem

lianzaozi, 2019-11-26:
When a dxf file is opened by the CAD software, use GRBL-Plotter software to open the dxf file again. GRBL-Plotter software will get stuck, so I recommend GRBL-Plotter to have a prompt message when opening any file that is being used .

[image]
Click the OK button in the figure above, the software is stuck. Opening the task manager to view the process of the plotter software also shows no response. This state persists, only the software is forcibly closed.
[image]

svenhb, 2019-11-26:
I will implement a check

svenhb, 2019-12-07:
Check is implemented, but could not be tested.

closed 2019-12-07 by svenhb