svenhb/GRBL-Plotter #88: Path planning;   the software adds the switch of the idle track display

lianzaozi, 2019-11-27:
[image]

I hope that the track display interface can be added to the right-click pop-up window to turn off the display and display of the idle track (green line in the figure), because sometimes there are more dxf file elements, there will be many green lines like this Will obstruct the red line and affect the observation.

In addition, can I let the user define which line is the first line when reading the dxf file, and can define which point of a line segment is the starting point. During path planning, user-defined line segments are planned according to user-defined rules, and the remaining users use the shortest path planning without defining them. Because in some laser processing, specific trajectories must be machined for pre-positioning, and then other trajectories are processed. The purpose of this is to prevent deformation during processing.

svenhb, 2019-11-27:
You can press the space-bar if the mouse is on the picture, to hide the "Pen-up" path.

To define the start point of the path is a lot of work, I don't have time for. Better do it in your drawing tool (I use Inksacpe, which can do that) or use other SW - Estlcam can do that.

lianzaozi, 2019-11-27:
Thank you very much
I found that the lines displayed in the method trace become thicker, but can't I change the displayed line width when zooming, just like CAD software. Is it possible to add offline trajectory simulation function, which can also simulate the movement of the trajectory without connecting the hardware

svenhb, 2019-12-07:
In GCode editor, selected figure can be moved up and down to change order

closed 2019-12-07 by svenhb