svenhb/GRBL-Plotter #89: Bug in importing svg when multile passes are enabled

majastanislawska, 2019-11-27:
It seems like when "use multiple passes" is used generated gcode falls after S0 command that disables laser.
This happens when importing SVG, gcode from 'generate simple shapes' works ok
this one is without mult8ple passes
```
(<Figure 2 Id="path4520" Color="#ff00ff" ToolNr="6"> )
G00 X5.98024 Y12.70002  (circle)
(Pen down: Z-Axis)
G01 Z-1.50000 F500 (circle Z values from tool table)
(Pen down: Laser-On)
S1000 (circle spindle speed from tool table)
G03 X5.98024 Y12.70002  I-1.48024 J0.00000 F800 (circle)
( SVG-Scale 0.788206875324249 1.26870250701904 )
(Pen up: Laser-Off)
S0 
(Pen up: Z-Axis)
G00 Z0.00000 ( Z feed from tool table)
(</Figure 2>)
```
and this one is with:
```
(<Figure 2 Id="path4520" Color="#ff00ff" ToolNr="6"> )
(Pen down: Z-Axis)
(Pen down: Laser-On)
S1000 (circle spindle speed from tool table)
( SVG-Scale 0.788206875324249 1.26870250701904 )
(Pen up: Laser-Off)
S0 
(<Pass 1 step='0.000' final='-1.500' >)
G00 X5.98024 Y12.70002  
G01 Z0.00000 F500 (  Z final.  Z feed  from tool table )
G03 X5.98024 Y12.70002  I-1.48024 J0.00000 F800 (circle)
G00 Z0.00000 
(</Pass1>)
(<Pass 2 step='-0.100' final='-1.500' >)
G00 X5.98024 Y12.70002  
G01 Z-0.10000 F500 (  Z final.  Z feed  from tool table )
G03 X5.98024 Y12.70002  I-1.48024 J0.00000 F800 (circle)
G00 Z0.00000 
(</Pass2>)
[..]
(</Figure 2>)```

Obviously effect is laser is off when machine is moving.

svenhb, 2019-11-28:
Thanks for investigating the problem, hope to fix it with the next version.

svenhb, 2019-12-07:
Bug fixed in version 1.3.2.4

closed 2019-12-07 by svenhb