svenhb/GRBL-Plotter #92: Without the buttons in your screenshot

lianzaozi, 2019-12-08:
[image]
Thank you very much for your reply
Without the buttons in your screenshot, the software version is V1.3.2.5。

When opening the dxf file through CAD software, please use GRBL-Plotter software to open the dxf file again. GRBL-Plotter software will be stuck, now after my test version V11.3.2.5 software will not get stuck when opening dxf occupied by other software, thank you

svenhb, 2019-12-08:
Strange... did you compile by yourself, or do you use the ready exe?

lianzaozi, 2019-12-08:
> 奇怪...是您自己编译的，还是使用现成的exe？

I unzip the file GRBL-Plotter_1324_publish.zip directly, and then run the GRBL-Plotter.exe program inside

lianzaozi, 2019-12-08:
I downloaded https://github.com/svenhb/GRBL-Plotter/releases on this link

svenhb, 2019-12-08:
Hopefully now it works:
https://github.com/svenhb/GRBL-Plotter/blob/master/GRBL-Plotter_1325_publish.zip

lianzaozi, 2019-12-09:
The problem is still the same, no button is displayed. 
Test environment: windows10 x64 home version version 1903

lianzaozi, 2019-12-09:
The problem has been solved, and the button is displayed after recompiling

svenhb, 2020-01-07:
Problem fixed

closed 2020-01-07 by svenhb